nodemcu/nodemcu-firmware #2277: sntp.sync fails even after Internet connection is established (in auto-repeat mode)

ritesht93, 2018-02-26:
------------------------ BUG REPORT -----------------------------------------
### Expected behavior
`sntp.sync` must perform sync successfully after Internet connection is established.

### Actual behavior
Here are the specific steps to reproduce:

1. Disconnect the Internet connection from your hotspot/router
2. Boot and connect ESP8266 to the router (This is important. ESP8266 must freshly boot and `sntp.sync` must fail on 1st attempt)
3. Start `sntp.sync` in auto-repeat mode
4. Reconnect the Internet after 2-3 failed attempts of `sntp.sync`
5. Actual Behavior: `sntp.sync` still keeps on failing

Note: the pattern of errors: error code `1` followed by `4` is necessary to reproduce the problem.

### Test code
Providing a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua

NodeMCU custom build by frightanic.com
	branch: master
	commit: 5073c199c01d4d7bbbcd0ae1f761ecc4687f7217
	SSL: false
	modules: file,gpio,net,node,rtctime,sntp,tmr,uart,wifi
 build 	built on: 2018-02-26 00:15
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
lua: cannot open init.lua
> =wifi.setmode(wifi.STATION)==wifi.STATION
true
> =wifi.sta.getip()
nil
> =wifi.sta.config({ssid="<ssid>", pwd="<password>"})
true
> =wifi.sta.connect()
> =wifi.sta.getip()
10.10.10.1	255.255.255.0	10.10.10.254
> sntp.sync(nil, function(sec) print(sec) end, function(err_code, err_desc) print(err_code, err_desc) end, 1)
> 1	0.nodemcu.pool.ntp.org
4	nil
4	nil
4	nil
4	nil
```

Note: after ESP printed `1`, `4`, `4`; the Internet connection was re-established and `sntp.sync` kept on failing with same error code.

### NodeMCU version
Branch: master
Revision: 5073c199c01d4d7bbbcd0ae1f761ecc4687f7217 (latest)

### Hardware
NodeMCU version 2 Amica

------------------------ END BUG REPORT -------------------------------------

FrankX0, 2018-02-26:
The autorepeat fails because of the initial DNS lookup failure. The current implementation will not try to recover from this automatically. So it will not retry a failed DNS lookup. After the initial attempt it assumes it correctly got a valid IP address.
There are several workarounds for this:

- Do not rely on the autorepeat functionality, but implement the retry (DNS lookup if needed) and re-sync with Lua code
- Avoid using DNS by using a fixed IP address of time server

Or we can have a look to fix this in the firmware, by applying the autorepeat also to the DNS lookup.

pjsg, 2018-02-26:
I'll take a look at this and fix it it properly.

pjsg, 2018-02-27:
I have a fix, but it needs more testing.... Hope to get a PR up in the next couple of days.

The current code is in https://github.com/pjsg/nodemcu-firmware/tree/fix-sntp-lookup

PR #2283 (2018-03-01): Fix SNTP so that it retries DNS lookups if it needs to
Fixes #2277

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

This retries the DNS lookups on each sync operation if the last sync operation ended in a timeout with no good responses, or if the dns lookups failed on the previous attempt. THis deals with the case when you start without a connection to th [... cut]
closed 2018-03-01 by marcelstoer