nodemcu/nodemcu-firmware #2278: NodeMCU/Linux development started [announcement]

Spiritdude, 2018-02-27:
As I wrote in https://github.com/nodemcu/nodemcu-firmware/issues/2252 I have been pondering on a general platform called "NodeMCU", where **NodeMCU/ESP8266** and **NodeMCU/ESP32** are two implementations - hence my interest to harmonize both API closely so the same Lua code can run on both.

Now, the past days I quickly patched together a skeleton for **NodeMCU/Linux** which utilizes the Linux environment to run NodeMCU Lua code, as a test case my NodeMCU Shell I currently work on; some **early** tests with just few partially complete modules you can check at https://spiritdude.wordpress.com/2018/02/26/nodemcu-linux/ (I will update that post frequently)

The project itself is hosted at (for now):
**NodeMCU/Linux https://github.com/Spiritdude/nodemcu-linux**

NodeMCU/Linux aims to support
- Raspberry Pi and RPi Zero running Raspbian
- NanoPi series with Allwinner H2 & H3 running Armbian
- Orange Pi series with Allwinner H2 & H3 running Armbian
- essentially any device which runs a Debian-based Linux distro

The idea is to implement majority of the base modules (to be defined clearly) in Lua itself, with few hooks with FFI (either `luajit` or `luaffifb` module). The main challenge will be modules like `u8g2` which needs low-level I2C or SPI interface hardware-near implemented.

If you are interested to contribute ideas or code, post an issue at https://github.com/Spiritdude/nodemcu-linux/issues (not here). If you have thoughts regarding the common platform (API) of "NodeMCU" as such, comment on this issue here. I currently document the "base" modules of NodeMCU/Linux at https://github.com/Spiritdude/nodemcu-linux/wiki - and the harmonization of NodeMCU/ESP8266 and NodeMCU/ESP32 https://github.com/nodemcu/nodemcu-firmware/wiki/ESP8266---ESP32-Compatibility

I'm only since ~2 months coding Lua, so I welcome any Lua wizard to give advise on good practice, e.g. see https://github.com/Spiritdude/nodemcu-linux/issues/1

Cc: @devsaurus @TerryE @marcelstoer

Spiritdude, 2018-03-05:
**First success** running **NodeMCU Shell** on **NodeMCU/Linux** (most crucial modules for it are working):

left-lower: **NodeMCU Shell** (telnet session)
right-upper: **NodeMCU/Linux** (starts a telnet server)

[image]

Most implemented modules haven't tested thoroughly yet, and will mature over the next weeks.

I currently use https://github.com/ignacio/LuaNode/ `luanode` where I adapted the `net.*` and wrote some small wrappers to adapt the NodeMCU API, which made `net.lua` quite small; also made `tmr.create():*` easier to implement as it has `setInterval()` (like in NodeJS).

`ffi` (luaffifb) works on Intel, but fails on ARM-based CPUs as my first tests have shown.

Current detailed state of the project https://spiritdude.wordpress.com/2018/02/26/nodemcu-linux/

TerryE, 2018-05-23:
I am not sure why this is remaining an open issue here.

closed 2018-05-23 by TerryE