nodemcu/nodemcu-firmware #2279: Exception raised on passing a callback to wifi.sta.connect and wifi.sta.disconnect

msurikov, 2018-02-27:
I am trying to pass callbacks to **wifi.sta.connect** and **wifi.sta.disconnect**.

### Expected behavior
The passed callbacks are invoked on WiFi connect/disconnect.

### Actual behavior
Exceptions are raised on executing **wifi.sta.connect** and **wifi.sta.disconnect** having callbacks (see test code in the next section). The exception messages are the following:
```text
stdin:1: bad argument #1 to 'connect' (number expected, got function)
stack traceback:
	[C]: in function 'connect'
	stdin:1: in main chunk
```
```text
stdin:1: bad argument #1 to 'disconnect' (number expected, got function)
stack traceback:
	[C]: in function 'disconnect'
	stdin:1: in main chunk
```
The exception messages are a bit confusing in stating that the first argument should be a number in [C] functions connect/disconnect. Actually, these exceptions are raised by **wifi.eventmon.register**(event, callback) which is being invoked by functions connect/disconnect and really expects the first parameter should be a number. Unfortunately, implementations of both functions wifi.sta.connect and wifi.sta.disconnect do pass three parameters (callback, event, callback) to wifi.eventmon.register and this is why we see the above exception messages. Definitely, the first parameter (callback) is odd here.

### Test code
```lua
wifi.sta.connect(function(T)
    print(string.format("WiFi connected to %s (%s) on channel %d.", T.SSID, T.BSSID, T.channel))
end)
```
```lua
wifi.sta.disconnect(function(T)
    print(string.format("WiFi disconnected from %s (%s) with reason %d.", T.SSID, T.BSSID, T.reason))
end)
```
### NodeMCU version
I use NodeMCU created by the cloud build service on 25-Feb-2018.

### Hardware
My ESP8266 device is Wemos D1 mini.

FrankX0, 2018-02-27:
Seems to be an issue in the current master build.
Please generate a dev build using the cloud service: works fine, where the master build fails as reported.

msurikov, 2018-02-28:
I've checked the Dev branch and can confirm - yes, this issue is already fixed there. Thanks a lot!

closed 2018-02-28 by msurikov