nodemcu/nodemcu-firmware #2280: gpio.PULLUP Not working (custom build)

chathurangawijetunge, 2018-02-28:
NodeMCU custom build by frightanic.com
	branch: dev
	commit: c8d2937fd55b1234ae8af7a84ae41560acebcdaf
	SSL: false
	modules: file,gpio,http,mqtt,net,node,rfswitch,tmr,wifi
 build 	built on: 2018-02-27 23:22
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
-----------------------------------------------------------------------------------------------
as of now the gpio.PULLUP are not working as earlier. is it a new bug..?  

 Status_led_Pin1 = 4  -- GPIO0 
 Status_led_Pin2 = 3  -- GPIO2 

gpio.mode(Status_led_Pin1, gpio.INPUT,gpio.PULLUP)  
gpio.mode(Status_led_Pin2, gpio.INPUT,gpio.PULLUP)
print(gpio.read(Status_led_Pin1),gpio.read(Status_led_Pin2))  ---> 0 ,0

pjsg, 2018-02-28:
Yes -- I introduced a bug with a recent PR...... I'll fix it....

chathurangawijetunge, 2018-02-28:
Thank will download customers build dev tomorrow

PR #2281 (2018-02-28): Fix problem with gpio.input
Fixes #2280

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

When I fixed the problem with gpio.mode not actually setting the output status, I (inadvertently) changed the behavior of gpio.input. This was due to a fall-through switch statement. My testing didn't catch it as interrupt inputs still worked.

This fix is somewhat ugl [... cut]
closed 2018-02-28 by marcelstoer