nodemcu/nodemcu-firmware #2294: getpeer() return do not reflect a valid TCP connection

devloperbra, 2018-03-09:
### Expected behavior
_net.socket:getpeer()_ should return ip and port, only when it´s connected, otherwise it should return _nil_, as explained on documentation:
[Returns](https://nodemcu.readthedocs.io/en/dev/en/modules/net/#returns_11)
port, ip (or nil, nil if not connected)

### Actual behavior
After execute a connection attempt to a TCP server, _net.socket:getpeer()_ returns same ip and port specified on net.socket:connect(port, ip).
net.socket:getpeer() return IP and PORT for arround5 seconds after net.socket:connect(port, ip) is executed.
After that, it returns nil,nil, as expected.

I would like to use getpeer() to determine if TCP is connected or not.

### Test code

```Lua
--CREDENCIAIS >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
--connect to Access Point (DO save config to flash)
station_cfg={}
station_cfg.ssid="xxxx"
station_cfg.pwd="xxxx"
station_cfg.save=false
--CREDENCIAIS <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

-- CONEXAO WIFI - INICIO >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

--print (station_cfg.ssid)

wifi.setmode(wifi.STATION)

wifi.sta.config(station_cfg)

wifi.sta.connect()

print("Adquirindo endereco IP...")

tmr.alarm(1, 500, tmr.ALARM_AUTO, function()
    ip = wifi.sta.getip()
    if ip ~= nil then
        print ("IP = "..ip)
        print("Conexao = ".. wifi.getphymode())
        print(wifi.PHYMODE_N)
        tmr.stop(1)
        print("Conectando ao servidor")
        TCP()--INICIALIZA O CLIENTE TCP / CONECTA AO SERVIDOR
    else
        print(".".."")
    end
end)--tmr.alarm(1, 500, tmr.ALARM_AUTO, function()

-- CONEXAO WIFI - FIM <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

--  TCP INICIO >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
function TCP()
  -- Cria cliente
  client = net.createConnection(net.TCP, 0)
  depurar_status_servidor()
  
  tmr.alarm(3, 1000, tmr.ALARM_AUTO, function()
    --testa se ja conectado
    porta = client:getpeer()
    if porta == nil then
      -- Connecta ao server
      client:connect(5001, "192.168.255.12")
      print(".".."")
    end --if porta == nil then
      
  end)

  -- Chama esta função ao receber dados no socket
  client:on("receive", function(socket, data)

    -- Simply print the received data
    print("TCP Rx = " .. data)
  end)      
  
  -- Cliente connectado ao server
  client:on("connection", function(sck,c)
    tmr.stop(3)
    -- Send some data to a server, for example
    client:send("Cliente TCP conectado")
    porta,ip = client:getaddr()
    print("(Local) IP:porta = ",ip,':',porta)
      
    porta,ip = client:getpeer()
    print("(Remoto)IP:porta = ",ip,':',porta)
    --client:close()
    UART_SETUP()--inicializa a UART
    
  end)--client:on("connection", function(sck,c)

--  TCP FIM <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
end --function TCP()
-- UART ----  INICIO >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
function UART_SETUP()
  uart.setup(0, 115200, 8, uart.PARITY_NONE, uart.STOPBITS_1, 0)
  print (uart.getconfig(0))
  uart.write(0, "Hello, world\n")

  uart.on("data", 1, function(data)
    print("receive from uart:", data)
    client:send(data)
        
    if data==";" then
      uart.on("data") -- unregister callback function
    end--if data==";" then
        
  end,0)--uart.on(
end --function UART_SETUP()
-- UART ----  FIM <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

function depurar_status_servidor()
  tmr.alarm(4, 1000, tmr.ALARM_AUTO, function()
      --testa se ja conectado
      porta,ip = client:getpeer()
      print("(Remoto)IP:porta = ",ip,':',porta)
    end)--tmr.alarm(4, 1000, tmr.ALARM_AUTO, function()
end
```
### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 5c8af3c452574448d471e271231b36ac3fff6b1b
	SSL: false
	modules: adc,bit,bme280,bmp085,dht,encoder,file,gpio,http,i2c,mdns,mqtt,net,node,pwm,sntp,spi,tmr,uart,ucg,websocket,wifi,wifi_monitor,ws2801
 build 	built on: 2018-03-06 20:43
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)

### Hardware
Lolin V3

TerryE, 2018-03-12:
> I would like to use getpeer() to determine if TCP is connected or not.

What you would like and what it does are two different issues.  This returns nil if the associated internal socket userdata `pcb`  isn't defined and the `remote_port` and `remote_ip` fields if it is.  These are initialised during socket connection and cleared when the userdata is garbage collected, which is not necessarily the same as when it is disconnected because of Lua GC algos.

closed 2018-03-12 by TerryE

devloperbra, 2018-03-12:
>  This returns nil if the associated internal socket userdata pcb isn't defined and the remote_port and remote_ip fields if it is

I see its working that way. I tested. It´s clear for me 

... but the doc says :
Returns
port, ip (or nil, **NIL IF NOT CONNECTED**)

Using your own words, "This returns nil if the associated internal socket userdata pcb isn't defined", what is different of docs, " **NIL IF NOT CONNECTED** "

TerryE, 2018-03-12:
@devloperbra, we all do this work pro bono.  Why don't you propose a change to the document wording and issue a PR?

devloperbra, 2018-03-12:
@TerryE I understand and appreciate your work. I work pro bono too on other job...

This is my first issue and very early development  stage with all these environment.

I am not sure where to propose this kind of PR.