nodemcu/nodemcu-firmware #2296: sntp dns lookups maybe incorrect behavior

pjsg, 2018-03-10:
### Expected behavior

I'm not sure -- I'm conflicted.

### Actual behavior

If you give a list of dns names to the sntp.sync method, then the call fails if *any* of them cannot be looked up. This seems bad. In my case, I had provided a couple of IP address and a couple of dns names, and the call failed because the internet was not reachable. [We got hit by the big storm in the north-eaast, and I was running on generator power for a couple of days and no internet.] My internal NTP servers continued to operate on generator power. 

### Solutions

# The current behavior is correct

# The sync operation should happen if *any* of the dns names or addresses can be converted into an ip address. There are then two sub cases: (a) at least one of the failed lookups should be reported on the error callback; (b) no error should be reported unless *all* of the lookups fails. 

I'm conflicted over which of the two subcases is better.

### NodeMCU version
dev

ritesht93, 2018-03-11:
I as the end user app developer would vouch for Solution 2, case b

pjsg, 2018-03-12:
I'm inclined to agree with you....

TerryE, 2018-03-15:
I vote for B also.

PR #2298 (2018-03-21): Ignore lookup errors on sync request unless all lookups fail
Fixes #2296.

Make sure all boxes are checked (add x inside the brackets) when you submit your contribution, remove this sentence before doing so.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This was largely driven by  [... cut]
closed 2018-03-21 by marcelstoer