nodemcu/nodemcu-firmware #2299: NodeMCU/* documentation

Spiritdude, 2018-03-15:
NodeMCU/Linux https://github.com/Spiritdude/nodemcu-linux moves forward at a steady pace, and I like to document things more detailed (e.g. different boards have specific features), but I wonder how to do it and what backend to choose.

In a bigger context I envision an unified NodeMCU documentation which documents all supported platforms (ESP8266, ESP32, Linux and others perhaps coming).

An idea would be, each platform write its documentation in a certain defined way (e.g. .md as it is now), and a script takes each section of all platforms and composes a large documentation (per module section):

----
## GPIO

This module provides access to the GPIO (General Purpose Input/Output) subsystem.

```
gpio.mode(pin,mode,...)
gpio.config({pin=p, mode=m, dir=d, ...})
gpio.write(pin,v)
gpio.read(pin)
gpio.trig(pin,...)
```

### gpio.mode

Initialize pin to GPIO mode, set the pin in/out direction, and optional internal weak pull-up.

#### Syntax
`gpio.mode(pin, mode [, pullup])`

#### Specifics
Feature | ESP8266 | ESP32 | Linux
--- | --- | --- | ----
`gpio.INPUT` | supported | supported | Raspberry: all GPIO supported, NanoPi NEO: ..., Orange Pi Zero: ...
`gpio.OUTPUT` | supported | supported | dito
`gpio.PULLDOWN` | ... | ...
`gpio.PULLUP` | ...
`gpio.PULLUPDOWN` | | | 
etc |

### gpio.config 

Initialize multiple pins to GPIO mode, set the pin in/out direction, and optional internal weak pull-up.

...
----

Important (for me) would be
- simplicity to write and maintain by developers
- consistency, encouraging current information (this is why I hint for automated gathering and unifying)

Any thoughts and recommendations?

Spiritdude, 2018-03-15:
The challenge is following (after further investigation):
- ESP8266 modules documentation:
  - come from different github repos (different developers)
  - are automatically incorporated into readthedocs.io

My thoughts right now go like this:
- using .md file format
- base documentation (based on current ESP8266 documentation) of base modules
  - removing ESP8266 specifics and moving those to ESP8266 specific documentation (see more detail below)
  - that process likely will speed up "NodeMCU API" definition
- each platform adapts:
  - `docs/`
    - `<language>/` (e.g. `en`, `fr`, `cn`)
      - `modules/`
          - `file.md`
          - `gpio.md`

with following logic:
- if doc file exists, then the module is supported, e.g. `file.md` or `gpio.md`
- the doc file only documents delta/difference/diverts/additional features of the base, e.g.
`nodemcu-linux/docs/en/modules/gpio.md`
  - if it's empty, then all features are support fully
  - if functions aren't supported, those must be listed ('##' entries) for example:
```
## gpio.trig()
- NanoPi NEO (with Armbian): currently triggers are not available.
- Orange Pi Series (with Armbian): latest Armbian support triggers.

## gpio.xyz_feature()
Not available.
```
which results in

----
# GPIO Module 
[base intro]
## gpio.mode()
[base info on gpio.mode()]
## gpio.config()
[base info on gpio.config()]
## gpio.trig()
[base info of gpio.trig()]

### Specifics
- NodeMCU/Linux: 
  - NanoPi NEO (with Armbian): currently triggers are not available.
  - Orange Pi Series (with Armbian): latest Armbian support triggers.

## gpio.xyz_feature()
[base info of gpio.xyz_feature()]

### Specifics
- NodeMCU/Linux: Not available

### Origin/Contributors
Platform | Origin/Contributor | Maintainer | Since
--- | --- | --- | ---
ESP8266 | Zeroday |  Zeroday | 2013/11/01
ESP32 | Johny Mattson | Johny Mattson | 2016/01/05
Linux | Rene K. Mueller | Rene K. Mueller | 2018/01/18

----

The actual rendering isn't that important (can be changed), important is the logic.

It would allow anyone to write a documentation for a (new) module, and other platforms have no matching file for it, hence, it's not yet supported (and indicated in the unified NodeMCU handbook):

Module | ESP8266 | ESP32 | Linux | Description
--- | --- | --- | ---- | ---
gpio | x | x | x | GPIO input / output
xyz | x | | | | something special 
raspberry_cam | | | x | VGA camera, Raspberry Pi series only

and module with different platform support looks like this:

----
# XYZ Module

Platform | Support
--- | ---
ESP8266 | full support
ESP32 | none
Linux | none

[base intro]

## xyz.init()

[base info xyz.init()]

-----

etc, you probably got the overall idea.

The main parsing of `docs/<lan>/modules/<module>.md` would be like this:
- '#' (single) is the title of the module, until first '##' the content is the intro of the module.
- '##' indicates the function description entry, those have to be coherent among the platforms.

Those of you with experience documentation on NodeMCU, let me know of oversights and challenges I likely have missed.

TerryE, 2018-03-15:
@Spiritdude Rene, in this sort of post it is really difficult to separate out the essence of the issue that you are rasing and the explanatory examples.  It is far readable if you move your long examples into gists, and link to them in your issue. 

But a couple of points:

1.  The publicly available documentation comes from a _single_ git repo -- _this_ one.  But yes we do accept PRs into our dev branch from other forks of our repository.  This is standard github practice.
2.  There is no _we_ when it comes to your project and this one.  Not as yet.  I don't know who you are or what your commitment and contribution credentials could be.  You have no karma on this project and that takes time and contributions to accumulate.    I feel that you have lots of drive and enthusiasm, but your project is only **25 days old**, and I have no idea as to whether in another 25 days you will have found some other project to take your time and so you move on.

So when it comes to an issue on _this project_, what I need you to explain (preferably in a few sentences per point: what the essence of the issue is as you see it; what you suggest is as a fix to the issue; what you feel that the benefits to _this_ project are; how much work is going to be involved; and (if you aren't going to provide all of the work, then how els edo you think will do this.

Spiritdude, 2018-03-16:
@TerryE 

1) OK.

2) right, I started NodeMCU/Linux as *my* repo, because the conceptual inertia here is stalling me and because I'm a newbie in so many ways. I simply didn't bother to discuss it here much further as I like to get things done - this is also why my changes in ESP32 I didn't compose as PRs yet, as I like to test them across the platforms and see what API is most suitable - I'm working at different fronts of "NodeMCU" as **a platform** which doesn't fit into a single github issue here . . .

So:
- what the essence of the issue is as you see it; 

Inconsistent API between ESP8266 and ESP32 due lack of a common documentation and vision.

- what you suggest as a fix to the issue; 

An unified documentation across platforms as a start.

- what you feel that the benefits to *this* project are; 

Simple maintainable documentation, eventually a consistent API across platforms.

- how much work is going to be involved; and (if you aren't going to provide all of the work then) who else do you think will do this.

I'm not confident that my technical proposal of an unified documentation is suitable yet, as I lack some of the technical details how you all maintain the documentation - so I first wanted to start the discussion of an unified documentation, and then (2nd post) to make a start how to do it. Once the technical details are agreed on - then I can estimate the overhead and say how much I will do - all of it I would prefer.

It seems to me the technical discussion is not possible here . . . I started yesterday with a script which merges the documentation, you can look at some results at https://github.com/Spiritdude/nodemcu-docs/tree/master/handbook/en/modules testing just two modules (`file` and `gpio`):
- used esp8266 docs as base (unedited)
- edited `esp32/file.md` and `esp32/gpio.md` which contains just the delta - see https://github.com/Spiritdude/nodemcu-docs/tree/master/docs/esp32/en/modules
- added `linux/file.md` and `gpio.md` to reflect missing features (not yet uploaded)

I continue to play and explore the best (least amount of work for everyone) way to compose this unified documentation. I will update this issue once I have done significant improvements. If anyone else has input and suggestions, please do voice it here.

TerryE, 2018-03-16:
> It seems to me the technical discussion is not possible here.

Why not? I welcome your input, and any constructive contribution.  All of our work and discussions are done through this repo and its issues list; everything is open here and available for mining, so it's quite a straightforward exercise to fill in the gaps in your understanding because the information is all here.

But one of the main constraints in this project is the limited time of those who are both technically able _and_ have the free time that they are willing to provide, pro-bono.  And even when we get both 'ticks in the box', those individuals tend to have their own priorities.  In my case for example, I have done the FAQ and sorted out a lot of internal constraints, such as my new LFS patch which effectively doubles or more the size of Lua applications that you can run on both ESP architectures.  My next major project is to finalise the Lua 5.3 port which both reunifies the core code-base for the two core ESP architectures and brings further memory and performance improvements. 

If you scan through past issues you will also see that both myself and Arnim have posted about standardising the APIs and library coding styles, and yes: this is the right thing to do; but no, we haven't been able to realistically resource it.  Also in previous issues, we have also discussed how we can unify the code and documentation trees of the two architectures.  So there have been a lot of ideas floated and discussed.

Ideas aren't the problem.  The issue is more one of coming up with the right proposals that have a realistic chance of being resourced.  

I would welcome your enthusiasm, but you seem not to have bothered to read in to project.  You haven't felt it necessary to introduce yourself to the rest of the project, in terms or your work, skill-set and levels of commitments, but at least making incremental improvements though the standard Github PR lifecycle, until we get to know you and your work.  Start the journey by taking small steps.  

Would your response have been any different if our roles were reversed?

**PS**.  Your Let's Encrypt HTTPS cert on your website has expired.

Spiritdude, 2018-03-16:
I talked with Arnim already and hope to talk to Marcel as well the next days . . . best we both talk online/phone the next days as well, it will clarify a lot.

devsaurus, 2018-03-18:
Let me add some thoughts on general concepts for technical documentation.

If I understood the proposal above correctly, then the goal for the result is to have a doc chapter per module which covers all variants esp8266, esp32, and eventually linux.
Such a chapter contains the superset of all variants with more or less ifs & else, exceptions and footnotes.

Pros of this approach:
- Commonalities are visible
- Overview on all functionality offered by the platform
- Differences are highlighted, potentially enabling further unification

Cons:
- Would a user of a particular platform be able to identify the content that applies to him?
- Would he be confused by n/a items?
- User has to parse the content in order to get what he's looking for
- ...

If I put myself into the position of a (new) user, then I'd like to learn from the docs what functionality is available and how to use it - with the particular board that's on my desk. I don't care about esp32 and linux if this board happens to be esp8266.
This "use case" of the docs would drive that we provide specialized documentation for each platform. Similar to what we have at the moment with dedicated manuals on RTD.

Pros of this approach:
- Content serves a single purpose: document a given platform. Nothing less, nothing more.
- Reviewing/maintaining the content just means to figure out whether it applies to the given platform.
- User finds what he needs

Cons:
- Risk of duplicating content which is the same across platforms
- Risk that similar content/functionality is not unified across platforms
- Hard to see how platforms differ since one would need to compare the module chapters
- ...

**TL;DR**
My opinion: If we intend to put the user into our focus then we should provide specialized documentation. We need to develop solutions that mitigate the risks which arise from this approach. Proposals are to be discussed.

Spiritdude, 2018-03-18:
@devsaurus great you join the *technical* and *strategic* discussion.

> If I put myself into the position of a (new) user, then I'd like to learn from the docs what functionality is available and how to use it - with the particular board that's on my desk. I don't care about esp32 and linux if this board happens to be esp8266.

You mean platform specific list of modules? I just added those:
https://github.com/Spiritdude/nodemcu-docs/tree/master/handbook/en

The list of functions with platform specific support as listed within the module descriptions can be rendered later (if I'm not constrained with .md) in a less invasive manner.