nodemcu/nodemcu-firmware #2300: Should we remove SQLite3 from dev?

TerryE, 2018-03-16:
### Issue

As discussed in #2297 the ***sqllite3** module compiles clean and works with trivial examples, but there is a major question as to where this will every work in consort with anything other than a trivial Lua example with the available ESP8266 RAM resources.  If it can't we used with a real Lua application, then why do we distribute this as part of dev?  Because if anyone does try to use it in anger and fails to get the application working, then we have no credible response to this sort of issue.

### Recommendation

-  Either the Original developer add a non-triival example of this libraries use to `lua_examples`, one that we can verify runs within the ESP8266 resources.
-  Or we backout this PR from dev

TerryE, 2018-06-23:
I go back to my original Q on this: has _anyone_ managed to get this working on a real application?  Or even weaker statement: Has anyone managed to use a firmware build with this module included?

Picking up your point @marcelstoer, if people have downloaded unusable firmware then we are really misdirecting potential users.

TerryE, 2019-04-24:
@marcelstoer I am sorting out the ROTable support and one the library modules that is going to need quite a bit of work is the sqllite3 one.  The original contributor hasn't been seen since this was committed.  The implementation notes for the library recommend a minimum 128Kb working RAM.  I know that you said that 6% of builds in the first few weeks after it was introduced included it, but we've had no issues raised over this last year despite the fact that it won't work in a real-life application, so I doubt that anyone has actually attempted to use it.

My vote is that we drop this library on the next commit.

NicolSpies, 2019-04-24:
On [https://db-engines.com/en/ranking](url) SQLite has moved up to 10 place out of 345. It has benefits as a self-contained, server-less, zero-configuration, relational database in embedded devices. It appears to be aiming for the edge of the network. My feeling is that the ESP32 might be a better fit. So I agree to drop this but place it in "file 13" for the future. Maybe it will find a home in the future.

TerryE, 2019-04-24:
@NicolSpies, I agree that SQLite is a very popular engine and there is a set of usecases where it is a fit. However, an IoT device with under 64K total free RAM is not one.

luizfeliperj, 2019-04-24:
> @marcelstoer I am sorting out the ROTable support and one the library modules that is going to need quite a bit of work is the sqllite3 one. The original contributor hasn't been seen since this was [...]

@TerryE I can update the library to cope with the new requirements, not a big deal. I haven't update it since so far because as far as I am aware (by being tagged) no trouble ticket about it has been opened.

Regarding the 128kb memory requirement, I can't find the source of this information. As far as I know it can be squeezed to fit on ESP8266 memory constraints.

If there is a demand to drop this library from the NodeMCU's repository, it should be because of other reasons, but not because it does not have a maintainer or due memory constraints.

marcelstoer, 2019-04-24:
I really don't see what the problem is. 
- Regardless of what you and I think of this module apparently there _are_ people who use it - or at least include it in their firmware. My stats alone say: 
  - included in 1656 builds (40 this month so far) by
  - 1126 individuals
- AFAIR no issue has ever been filed against this module. It either just works or it's not used but then why build it into the firmware in the first place?
- Its maintainer is willing to look after it and make adjustments if necessary. Who responses in less than 1 day to comments about his module on this issue list.

> but we've had no issues raised over this last year

If we were to apply this metric consistently we'd have to drop 3/4 of our modules.

> despite the fact that it won't work in a real-life application

I never tried myself personally so I can't say. However, I find that claim a bit unfair. (As I said I don't know if it's well founded or not but) if someone else made such a claim about code you wrote you would ask for proof in the form of a small test application, no?
Of course it depends on ones personal definition of _real-life application_. To me it doesn't matter much actually. Even if everybody who uses it built Mickey Mouse applications...if the module proves to be useful for them so be it. Precondition of course is that maintenance effort is zero or shouldered by the contributor or someone else.

TerryE, 2019-04-24:
@luizfeliperj @marcelstoer, In raising this issue my original suggestion had two options: add a non-triival example of this libraries use to `lua_examples` that we can verify runs within the ESP8266 resources; _**or**_  we backout this PR from dev.  

13 months has passed and we _still_ don't have the example.  The sqlite3 documentation suggests [here](https://www.sqlite.org/malloc.html) that typically 40Kb RAM is needed for sqlite buffers.  Even if our entire application is running out of LFS, we still need RAM for the application, SPIFFS and network buffers, etc., so I just can't see how a real-world application will run successfully on the ESP8266.  So am I being unfair?

Luiz, have you used the library module in anger yourself yet -- that is with a real Lua application that is connecting to IoT devices using MQTT for example?  You certainly don't seem to have any such example in your GitHub repo.

TerryE, 2019-10-28:
@luizfeliperj @NicolSpies @marcelstoer @Sledmine, I note that another six months has passed and still no example of anyone successfully using this library.  Why keep a module in our distro with the associated maintenance costs and trigger issues like #2913 when we have seen zero examples of its being used?

SQLite3 does make sense for the ESP32 which meets the minimum RAM requirements, but not the ESP8266.  In my view we should drop this module from our ESP8266 distros.

NicolSpies, 2019-10-28:
@TerryE , a small local relational database is still on my list to be used in future. With the current rebasing that will potentiate the ESP32, I concede to using SQLite3 on the ESP32 in the future.  :+1:

TerryE, 2019-10-28:
@NicolSpies if you do, can I suggest that you might be better going back to a more standard SQLite3 installation rather than this shoe-in.

luizfeliperj, 2019-10-28:
@TerryE, as I wrote in #2913 (comment 547005633), I was unaware that the suggested API changes (as of comment 486029042) already took place.

In regarding the heap usage, it is a trend that every advance in the SDK version the less will be the heap available. This way, in the future, all modules will break. So what will happen? All modules will be dropped out or silently '#if 0'ed?

Moreover, by taking a deeper look at the defines, one can figure out that adjusting SQLITE_DEFAULT_PAGE_SIZE will reduce the heap requirements of the library at the cost of breaking compatibility with the default library settings. Depending on the size of the database and what type of operation, it will be required 2 or more SQLITE_DEFAULT_PAGE_SIZE of heap to get the database modified. In short, less the heap will lead to less complex database operations and/or the usage of smaller page size, this is not an unsolvable problem.

Other than that, if one is using WiFi and needs to make a query, disconnecting from the WiFi to get more heap and then reactivating WiFi is an perfectly suited workaround, the problem that needs to solve will define the requirements, the same workaround can be done to save on energy. The module still has its uses even with such restrictions.

Summarizing, I believe this is a never ending discussion. I was not expecting to be received by the NodeMCU community this way. I contributed the sqlite3 module code because of a project I had in mind and I, at least for now, can't afford spending more time with that. There are people out there that might want to use it and I believe this excuse is enough to have this code available, this is so much true that once the library was forcefully broken someone raised an issue, but interestingly there are no issues regarding the library usage or heap problems when using it.

I don't think it is worth spending any more energy and/or time on this issue, so, please close this issue with whichever decision you may have in mind.

TerryE, 2019-10-28:
@luizfeliperj, getting SQLite3 to build on an ESP8266 target within NodeMCU was a challenge, but what I've asked the community to provide, and you as the original contributor included, is an example real application that uses it. No one has stepped forward with a response to that Q, and that is why I am unwilling to put my time into maintaining it. 

If you also don't want to put in that time, then I think that there is only one consensus outcome.

TerryE, 2019-10-29:
@luizfeliperj, whilst the SDK is slowly getting bigger, when I boot the current Lua 5.3 alpha build with my app in LFS, I have 44Kb RAM available and since the TValue size is now 8 bytes, this represents more data variable available than we've ever had before.  

In my view, it really was a mistake to try to shoe-horn SQLite3 into an ESP8266 build, and I still don't see it as usable in a real-world appliication.  In the ~2 years since you first did this work to my knowledge, we haven't had _a single_ contributor -- and that includes you as the OC -- come forward with such an application that uses it.

I had to go through all of the C modules to unify the calling APIs to work across both Lua 5.1 and the Lua 5.3 APIs which was mostly an exercise of lots of scripts and bulk edits.  My challenge with SQLite3 was that there was going to be as much work involved with getting this one module to work and to be able to test it than for all of the other modules combined.  I just didn't have the bandwidth to do this, and given that we've got no current users, it was just easier to `#if 0` it out for now than to stall the migration.

Where SQLite3 would be a far better fit to processing and RAM resources is our ESP32 variant, and moreover in these last 2 years these modules are now easily obtainable.  So my suggestion would be to target the ESP32 and not the ESP8266, but if you do so then _please_ use a more standard library build rather than using a single module some **202,286** C lines long.

closed 2020-06-10 by marcelstoer