nodemcu/nodemcu-firmware #2304: sigma_delta code is always included even when not needed

pjsg, 2018-03-17:
### Expected behavior
The sigma_delta code should only be included if needed

### Actual behavior
It always occupies space in the iram

### Root cause
`app/platform/platform.c` (which is always required) consumes IRAM for sigma_delta. Those few functions should be split out into their own C file.

In my environment, this allows the GDBSTUB to be included and the modules that I need.

TerryE, 2018-03-18:
Phillip, this is a wider issue than `sigma_delta`.  The general rule should be:

-  _Any _ modules which uses the `ICACHE_RAM_ATTR` should only emit code / data if the selected module configuration requires it.  

Because we link with the necessary options to remove unused modules, in general this only relates to code embedded in mandatory modules such as `platform.c`, but this is also the H/W timer code which is only need by gpio.serout, gpio.pulse, perf and somfy

TerryE, 2018-03-24:
> It always occupies space in the IRAM

I can't recall how much this uses but it is perhaps 100s of bytes.  Pondering this one a bit more, the  _real_ issue isn't the 100s of bytes, it is the fact that the IRAM resource isn't well defined: use too much and your firnware get stuck in a reboot loop spraying out error messages before the UART bard rate is 115200 so this comes out as garbage;  shave off another few 100 bytes and everything works perfectly.  But I don't know of any documented well-defined way of probing a mapfile to know if any individual firmware image will boot or barf.

A current example is debugging Petr's (@joysfera) latest issues #2311. When I added the remote gdb stub this kicked the image over this limit: reboot and crap forever,  so I had to go through the cose and remove all of the libraries that this test case didn't need, and bang -- now it loads.

devsaurus, 2018-03-24:
> `app/platform/platform.c` (which is always required) consumes IRAM for sigma_delta. Those few functions should be split out into their own C file.

I agree that this is not desirable, but don't we look at a more fundamental issue here than just the question whether files and functions are partitioned in the optimum way?

The model I have in mind expects things to work roughly like this (maybe naive):
1. C code is compiled with `-ffunction-sections`
2. Linking stage gets `-Wl,--gc-sections` and will omit all unused function sections from the output
3. If sigma_delta is not enabled then the dependency tree analysis will remove all related functions

**But** there are traces of sigma_delta in the image. Why?
```
$ xtensa-lx106-elf-objdump -d app/.output/eagle/debug/image/eagle.app.v6.out | grep sigma_delta
4010666c <platform_sigma_delta_set_target>:
40106678:       000b05          call0   4010672c <sigma_delta_set_prescale_target>
40106695:       fffbc1          l32r    a12, 40106684 <platform_sigma_delta_set_target+0x18>
401066a5:       fff841          l32r    a4, 40106688 <platform_sigma_delta_set_target+0x1c>
401066e6:       ffe921          l32r    a2, 4010668c <platform_sigma_delta_set_target+0x20>
40106705:       ffe121          l32r    a2, 4010668c <platform_sigma_delta_set_target+0x20>
4010672c <sigma_delta_set_prescale_target>:
40106778:       fffe21          l32r    a2, 40106770 <sigma_delta_set_prescale_target+0x44>
40106784:       fffc21          l32r    a2, 40106774 <sigma_delta_set_prescale_target+0x48>
40106794:       fff721          l32r    a2, 40106770 <sigma_delta_set_prescale_target+0x44>
401067a0:       fff521          l32r    a2, 40106774 <sigma_delta_set_prescale_target+0x48>
40106869:       ffc131          l32r    a3, 40106770 <sigma_delta_set_prescale_target+0x44>
```

It seems that the usage of `ICACHE_RAM_ATTR` breaks the above garbage collection mechanism already in the `-ffunction-sections` step. Functions with this macro are placed into the default `.iram0.text` section instead of a specialized `.iram0.text.whatever` section - the linker has no chance to remove the unused functions.

Can we improve `ICACHE_RAM_ATTR` to be compatible with the `--gc-sections` approach?

TerryE, 2018-03-24:
The uses of IRAM code in to core of lua + platform are so small that IMO, we should just add the relevant `#ifdef` code variants to ensure that these platform components just return a flash-based null / nill return is the corresponding option isn't configured.

pjsg, 2018-03-25:
The (arguably) easier way is to add:
```
#define RAM_ATTR(fn) __attribute__((section(".iram0.text." #fn))) fn
```

and then make the following change:

```
-void ICACHE_RAM_ATTR platform_sigma_delta_set_target( uint8_t target )
+void RAM_ATTR(platform_sigma_delta_set_target)( uint8_t target )
```

This is scalable to all the ICACHE_RAM_ATTR functions if we think that they could be omitted.

devsaurus, 2018-03-25:
So we have three options?
1. Move `platform_sigma_delta_*` to a separate file
2. Apply the section trick to enable gc-sections on IRAM
3. Enclose code with `#ifdef`

Option 1 addresses this issue, but we might get struck again in the future if something similar slips through review.
Option 2 would also work, though I'd propose to change `ICACHE_RAM_ATTR` to support section names and maybe have a specialized variant for instances where we don't want this. Pro: future additions with ICACHE_RAM_ATTR will be compatible from the start. Con: Needs changes in a number of locations while the issue is just relevant for `platform_sigma_delta*`
Option 3 requires to hardcode the dependencies. Is feasible, but requires manual maintenance for future additions.

pjsg, 2018-03-25:
Actually, I wonder if 
```
#define STRINGIZE(x) STRINGIZE2(x)
#define STRINGIZE2(x) #x
#define LINE_STRING STRINGIZE(__LINE__)
#define ICACHE_RAM_ATTR     __attribute__((section(".ram.text." __FILE__ "." LINE_STRING)))
```

will work?

devsaurus, 2018-03-26:
Smart trick, Philip!

It just works:
```
$ xtensa-lx106-elf-objdump -d app/platform/.output/eagle/debug/lib/libplatform.a | grep Disassembly
Disassembly of section .ram.text.hw_timer.c.98:
Disassembly of section .ram.text.hw_timer.c.105:
Disassembly of section .ram.text.hw_timer.c.53:
Disassembly of section .ram.text.hw_timer.c.73:
Disassembly of section .text.platform_hw_timer_set_func:
Disassembly of section .ram.text.hw_timer.c.118:
Disassembly of section .text.platform_hw_timer_init:
Disassembly of section .ram.text.hw_timer.c.166:
Disassembly of section .text.set_gpio_no_interrupt:
Disassembly of section .ram.text.platform.c.211:
...
```
When I change l.124 in `ld/nodemcu.ld` to
```
*(.iram.text .iram0.text .ram.text.*)
```
then all ends up nicely in the image.
This is the result from a build with default module selection:
```
$ xtensa-lx106-elf-objdump -d app/.output/eagle/debug/image/eagle.app.v6.out | grep -E 'Disassembly|<.*>:' | less
isassembly of section .text:
40100000 <_stext>:
40100004 <call_user_start>:
40100010 <_DebugExceptionVector>:
40100020 <_NMIExceptionVector>:
40100030 <_KernelExceptionVector>:
40100050 <_UserExceptionVector>:
40100070 <_DoubleExceptionVector>:
4010009c <_NMILevelVector>:
4010015c <_UserExceptionVector_1>:
401001b0 <Cache_Read_Enable_New>:
401001ec <NmiTimSetFunc>:
40100214 <NMI_Handler>:
4010026c <call_user_start_local>:
...
40106278 <user_start_trampoline>:
40106388 <load_non_32_wide_handler>:
4010641c <__wrap__xtos_set_exception_handler>:
40106458 <uart0_rx_intr_handler>:
40106548 <platform_gpio_intr_dispatcher>:
40106678 <seroutasync_cb>:
40106714 <hw_timer_isr_cb>:
40106730 <hw_timer_nmi_cb>:
40106750 <platform_hw_timer_arm_us>:
40106780 <platform_hw_timer_close>:
Disassembly of section .irom0.text:
40210000 <uart_div_modify-0x14>:
40210014 <uart_div_modify>:
...
```

pjsg, 2018-03-30:
I'll prepare a PR with this in it....

PR #2334 (2018-04-07): Put each ICACHE_RAM_ATTR function in its own section.
Fixes #2304.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

This implements putting each ICACHE_RAM_ATTR function in its own section. This allows the linker to throw away more functions and hence use less ram.

I checked the map and it looks good. The image boots and seems to work.
closed 2018-04-08 by marcelstoer