nodemcu/nodemcu-firmware #2306: uart.on: wrong arg range

ZaXk, 2018-03-18:
From page: https://nodemcu.readthedocs.io/en/dev-esp32/en/modules/uart/#uartstart

Syntax
uart.on([id], method, [number/end_char], [function], [run_input])

When I run this code:
`uart.on(1,"data", "\r",
  function(data)
    print("receive from uart:", data)
    if data=="quit" then
      uart.on("data") -- unregister callback function
    end
end, 0)`

error output:

> uart.lua:4: wrong arg range
stack traceback:
	[C]: in function 'on'
	uart.lua:4: in main chunk
	[C]: in function 'dofile'
	stdin:1: in main chunk

FrankX0, 2018-03-20:
Unlike the documentation suggests, `uart.on()` does not support an (optional) uart id number.

ZaXk, 2018-03-23:
It's my fault. It's esp32 doc. My board is esp8266.

devsaurus, 2018-03-23:
Doc confirms that on ESP32 `uart` module has an additional optional input parameter which is not present on ESP8266 API.

closed 2018-03-23 by devsaurus