nodemcu/nodemcu-firmware #2308: MQTT subscribe payload max size severely limited

joysfera, 2018-03-19:
### Expected behavior
NodeMCU should receive every message that has been published to a MQTT topic it is subscribed to.

### Actual behavior
If the message or payload is longer than 984 bytes (in my tests with 35 bytes long topic) no message is received.
Perhaps the actual limit is the sum of topic length and payload/message length but it is still below 1 kB which is a severe limit. FYI, the payload max size is 256 MB per MQTT spec. Of course I don't want an ESP device with 20 kB free heap to process 256 MB message but a hidden/undocumented limit to less than 1000 bytes is not fun either. Worst is that no message is received at all, there's no indication that a message got lost. If it at least returned first 1, 2 or better yet 4 kB of the message...

### Test code
```Lua
mqtt_client = mqtt.Client(1, 120, name, passwd)
mqtt_client:on("message", function(client, topic, data) print(#data) end)
mqtt_client:connect(broker, 1883, 0, 0, function(client)  mqtt_client:subscribe({["#"]=1}) end)
```
### NodeMCU version
current master

### Hardware
plain ESP8266-07

joysfera, 2018-06-23:
Looking into app/modules/mqtt.c - this is suspicious:

```
#define MQTT_BUF_SIZE 1024

READPACKET:
  if(length > MQTT_BUF_SIZE || length <= 0)
          return;
```

This `MQTT_BUF_SIZE` define is used at six places (received, connected, timer, unsubscribe, subscribe, publish) to allocate a temporary buffer **on the stack**.

I wonder if it was safe to increase the `MQTT_BUF_SIZE` to say 2048 or 4096? The 1kB limit on topic+payload is too severe and handling of messages larger than that is non-existent - they're just ignored on both publish and receive. Scary!

How much space there's left on stack normally? Wouldn't it cause random and hard to debug issues elsewhere?

joysfera, 2018-07-24:
Weird things are happening there! I have just compiled my own firmware from the master branch and increased the `MQTT_BUF_SIZE` from 1024 to **2048** there. I was expecting to be able to receive packets up to 2042 bytes then. Imagine my surprise when I was able to receive packets longer than 10 kB, i.e. much larger than the `MQTT_BUF_SIZE`! Granted, the content of the message (the payload) is cut at 2042 bytes but larger messages do NOT get lost, contrary to the firmware I've been using.

I have quickly flashed back the "official" build (from nodemcu-build.com) to ensure that I am indeed stuck with 1018 bytes max (using the `printf` test outlined above). Indeed, MQTT with 1019+ bytes disappear (are not received by NodeMCU). Does it mean that the MQTT is broken in the "official build" while my own build can receive even large the MQTT packets properly?

BTW, what does @TerryE think about increasing the MQTT_BUF_SIZE?

joysfera, 2018-07-25:
Understand. I just thought you'd know from top of your head what stack size is available in NodeMCU and if it is safe at all to allocate 1kB - 2kB buffers on the stack.

joysfera, 2018-07-29:
I have built several more configurations to find out what MQTT buffer size is necessary to stop losing MQTT packets. 1280 bytes is not enough. 1536 bytes is OK. So it seems to be pointing at TCP packet size. If the TCP packet fits to the MQTT buffer completely then the MQTT communication works even for much larger packets. However, if the first TCP packet does not fit to MQTT buffer then the message is not received at all.

The MQTT implementation in NodeMCU is in rather bad shape in many ways but if it was possible to increase the MQTT working buffer on stack from 1024 to 1536 bytes it would (at least partially) improved larger MQTT messages handling.

stromnet, 2018-11-02:
Hi, 

had some issues with small but rapidly sent messages getting dropped by my ESP12-F on NodeMCU 2.2.1.
The messages where just 52 bytes long (33 byte topic), but sent with 15 consecutive calls to `mosquitto_pub` (so at most some shell & connect overhead between msgs). Problem was that I was just getting the first one or two messages. 

Raising `MQTT_BUF_SIZE` to 1536 removed this problem, which gives additional weight to the suggestion above that we are dealing with TCP packet size limitations.

Just my 2 cents, if it can help anyone :)

closed 2018-11-13 by marcelstoer

stromnet, 2018-11-30:
@joysfera do you have the possibility to test out the #2571 fix, to see if this fixes the issues you are seeing properly?

joysfera, 2018-11-30:
> @joysfera do you have the possibility to test out the #2571 fix, to see if this fixes the issues you are seeing properly?

I have published a test code above in the issue description so anyone can test it. I myself have been using my own patched version for four months already but I went for **1536** bytes because I like round numbers :-) I will test your 1500 bytes version when I get a chance.

stromnet, 2018-11-30:
#2571 is a proper fix which does not rely on a specific local static buffer, and handles slot more scenarios where old code (regardless of buffer size) would fail.
See exact details in PR/commits :)

joysfera, 2018-11-30:
Ah, sorry then, I remember seeing somewhere a 1500 bytes buffer PR so I commented on that one. Will definitely check the proper fix because MQTT has (or used to have) many weird issues in NodeMCU. Some of them might got fixed but #2571, hopefully?

stromnet, 2018-11-30:
Right, my first fix was just changed size, then got around to fix it properly!

That might very well be the case, if it was related to dropped/missing messages at least. Quite a few cases which resulted in that..

joysfera, 2018-12-25:
@stromnet OK, got to testing this, at last. The #2571 does NOT fix this issue. My test case in the original description fails for 1019 character long message.

stromnet, 2018-12-25:
@joysfera, if you where using the exact same code then I would expect it to fail. If you do not add the `max_message_length` param to the mqtt.Client constructor, it will default to 1024 (for full message, including headers & topic name).

It should not have been delivered to the regular `.on('message', ..` callback though, but rather to a `overflow` callback. Was it actually given to `message` callback?

joysfera, 2018-12-25:
Damn, after rebooting the ESP8266 and re-trying identical set of commands (well, only changed nil for '') it started working correctly. Now the on("message") receives messages up to 1530 bytes (for 1.5kB `max_message_length`). So everything is awsome, thank you! :)

joysfera, 2018-12-25:
Now I just have to try out if the older `Client()` can handle an additional parameter without going crazy.

stromnet, 2018-12-25:
Great!

Yes, older Client should just ignore it. 
Calling `client.on('overflow'` on a older version will raise a LUA error though, `method not supported`.

PR #2544 (2018-11-13): Increase MQTT recv buffer size to support MTU 1500  (#2308)
Any TCP packet with more than 1024 bytes of payload was silently
dropped. With MTU of 1500 the TCP payload can be up to 1460 bytes
(1500 - 20(IP hdr) - 20(TCP hdr))

Fixes #2308

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution... **But I'm quite a new NodeMCU & MQTT user so I might be unaware of som [... cut]