nodemcu/nodemcu-firmware #2313: DS18b20 integer readout: decimals wrong for negative temperature

peturdainn, 2018-03-23:
While calibrating DS18b20 using ice-water, I noticed readouts flipping from 0 to 0.9 degrees C. 

Upon checking /app/modules/ds18b20.c I see this calculation for the decimals:

ds18b20_device_scratchpad_temp_dec = ((double)(ds18b20_device_scratchpad[0] & 0x0F) / 16 * 1000);

Which does not take into account the sign bit(s) from scratchpad[1] so negative decimals are calculated as positive

ref datasheet:
https://datasheets.maximintegrated.com/en/ds/DS18B20.pdf page 6

I'll see if I find time to come up with a better function

devsaurus, 2018-03-23:
We've had a similar issue with the ads1115 in #2207. Don't know if the particular problem you're reporting here is related, but the topic of wrong sign handling around 0 point in integer builds might be present in `ds18b20` module as well.
Any PR is highly appreciated :smiley:

peturdainn, 2018-03-23:
I had a quick look at the ads1115 code and it certainly isn't related codewise, I'd need to look at the datasheet to see if it's related wrt root-cause.

The root-cause of this one here is that the data is 16bit read split over two registers, to highest one is used for the integral part, and since it contains the sign bit the result is ok. 
The decimal part comes from the lower byte, which doesn't contain any trace of the sign bit and as such is always interpreted as a positive number.
So temp goes from 0.0 to -0.9

The fix is easy but I have been a lazy user of the web build service (there's only so many stuff you can work on in detail), so I'm setting up a build environment now ;)

peturdainn, 2018-03-25:
PR done, I checked the contributing doc so I hope it checks out fine, shoot if there's an issue ;)

peturdainn, 2018-03-26:
I think it shows I'm kinda new to creating PR on Github ;)

Sorry about the noise

TerryE, 2018-04-23:
Sorry, I've just posted a comment on the PR that really belongs here on the issue so I am reposting it here.

I've just taken a look at this code and some of the logic here is plain bizarre. I am talking about the original module, though @peturdainn's mods don't help. 

-   First, I would have expected some conditional logic based on `LUA_NUMBER_INTEGRAL`.  The CB should  only return a single value for floating point builds.
-   In the case of integer build, yes it might make sense to return an integral part and a decimal fraction part, but what these are is not well defined, and this confusion lies at the heart of this issue / PR. 
-  whether or not, the build is integral, internally within the module the calculations should be done in floating point. 

So taking an example what should a reading of - 1.4°C return on integer builds?
-   The number of decimal places isn't explicitly defined or documented. Should this be API configurable, or fixed at 3?  Let's assume 3 for the following.
-   - 2, 600 - valid and the maths convention but not easy to use for Lua programmers.
-   - 1, 400 - definitely not because 1000*(-1)+400 is the **wrong** answer.
-   - 1, - 400 - this is the one that makes the most sense to me. This is also easy to use in printing so long as you use `maths.abs() ` on the decimal component.

This PR doesn't do this so :-1:  from me.

TerryE, 2018-04-23:
> I don't see how **1, -400** is a good representation for -1.4.  The patch makes it return **-1, -400** ... before the patch it would return **-1, 600**

Sorry @peturdainn, I misread your code because the implementation is a bit tangled.  It's sounds like we came to the same conclusion.  This also satisfies @pjsg's linearity test.

PR #2322 (2018-04-27): fix for ds18b20 negative decimals
Fixes ds18b20 negative decimals bug by changing the way temp and temp_dec are calculated, taking care not to lose the sign of the decimals part

Fixes #2313

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

ds18b20 decimals do not take into account the sign bit. Since the original calculation was not so readable, rewritten  [... cut]
closed 2018-04-27 by marcelstoer