nodemcu/nodemcu-firmware #2317: Boot loop at attempting to create file system - ESP32

zortech, 2018-03-25:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
File System is detected as missing, file system is created.

### Actual behavior
entry 0x40078fd8
I (30) boot: ESP-IDF v3.1-dev-562-g84788230 2nd stage bootloader
I (30) boot: compile time 23:49:43
I (30) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x19bc0 (105408) map
I (132) esp_image: segment 1: paddr=0x00029be8 vaddr=0x3ffb0000 size=0x01da4 (  7588) load
I (135) esp_image: segment 2: paddr=0x0002b994 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/zor/esp/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (139) esp_image: segment 3: paddr=0x0002bd9c vaddr=0x40080400 size=0x04274 ( 17012) load
I (154) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x80480 (525440) map
0x400d0018: _stext at ??:?

I (339) esp_image: segment 5: paddr=0x000b04a0 vaddr=0x40084674 size=0x0be90 ( 48784) load
0x40084674: i2c_master_cmd_begin_static at /home/zor/esp/esp-idf/components/driver/./i2c.c:947

I (359) esp_image: segment 6: paddr=0x000bc338 vaddr=0x400c0000 size=0x00064 (   100) load
I (369) boot: Loaded app from partition at offset 0x10000
I (369) boot: Disabling RNG early entropy source...
I (371) cpu_start: Pro cpu up.
I (374) cpu_start: Starting app cpu, entry point is 0x40081328
0x40081328: call_start_cpu1 at /home/zor/esp/esp-idf/components/esp32/./cpu_start.c:225

I (366) cpu_start: App cpu up.
I (385) heap_init: Initializing. RAM available for dynamic allocation:
I (392) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (398) heap_init: At 3FFB8310 len 00027CF0 (159 KiB): DRAM
I (404) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (410) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (417) heap_init: At 40090504 len 0000FAFC (62 KiB): IRAM
I (423) cpu_start: Pro cpu start user code
I (105) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

Mounting flash filesystem...
No filesystem partition found, attempting to create it...
abort() was called at PC 0x400eaccd on core 0
0x400eaccd: is_safe_write_address at /home/zor/esp/esp-idf/components/spi_flash/./flash_ops.c:113

Backtrace: 0x4008dcf0:0x3ffbb080 0x4008dec7:0x3ffbb0a0 0x400eaccd:0x3ffbb0c0 0x400836c7:0x3ffbb0e0 0x40144e06:0x3ffbb100 0x400eae49:0x3ffbb120 0x400eae88:0x3ffbb160 0x400eb19d:0x3ffbb1b0 0x400eb1a7:0x3ffbb1d0 0x400ea862:0x3ffbb1f0 0x400f0c87:0x3ffbb220 0x400f0cec:0x3ffbb240 0x400d1736:0x3ffbb280
0x4008dcf0: invoke_abort at /home/zor/esp/esp-idf/components/esp32/./panic.c:648

0x4008dec7: abort at /home/zor/esp/esp-idf/components/esp32/./panic.c:648

0x400eaccd: is_safe_write_address at /home/zor/esp/esp-idf/components/spi_flash/./flash_ops.c:113

0x400836c7: spi_flash_erase_sector at /home/zor/esp/esp-idf/components/spi_flash/./flash_ops.c:196

0x40144e06: platform_partition_add at /home/zor/esp/nodemcu-firmware-esp32/components/platform/./platform_partition.c:97

0x400eae49: get_spiffs_partition at /home/zor/esp/nodemcu-firmware-esp32/components/spiffs/./spiffs.c:135

0x400eae88: myspiffs_mount_internal at /home/zor/esp/nodemcu-firmware-esp32/components/spiffs/./spiffs.c:135

0x400eb19d: myspiffs_mount at /home/zor/esp/nodemcu-firmware-esp32/components/spiffs/./spiffs.c:135

0x400eb1a7: myspiffs_vfs_mount at /home/zor/esp/nodemcu-firmware-esp32/components/spiffs/./spiffs.c:135

0x400ea862: vfs_mount at /home/zor/esp/nodemcu-firmware-esp32/components/platform/./vfs.c:421

0x400f0c87: nodemcu_init at /home/zor/esp/nodemcu-firmware-esp32/components/base_nodemcu/./user_main.c:134

0x400f0cec: app_main at /home/zor/esp/nodemcu-firmware-esp32/components/base_nodemcu/./user_main.c:169

0x400d1736: main_task at /home/zor/esp/esp-idf/components/esp32/./cpu_start.c:467

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
make defconfig
make
make flash

### NodeMCU version
dev-esp32

### Hardware
HiLetgo ESP-WROOM-32 

8<------------------------ END BUG REPORT -------------------------------------

devsaurus, 2018-03-28:
This is a symptom when the latest ESP-IDF is included instead of the version that's configured as submodule. Please ensure that your clone includes https://github.com/espressif/esp-idf/tree/9a26296a0e88a4c3ae27e9c848be970946fff87e.

devsaurus, 2018-04-04:
I've updated the `dev-esp32` branch to latest IDF, the firmware avoids this bootloop now. The new problematic config entry was `CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS` which is now overruled in `sdkconfig.defaults`.

Since you already included IDF 3.x as submodule, I recommend you remove the existing `sdkconfig` and set up the firmware again.

devsaurus, 2018-04-15:
Issue considered fixed.

closed 2018-04-15 by devsaurus