nodemcu/nodemcu-firmware #2321: net socket documentation clarification

cwrseck, 2018-03-26:
The Lua Developer FAQ has the following example of socket usage:
> srv:on("connection", function(sck, c)
>     svr:send(reply)
> end)

As far as I can tell from the wording this is intended as an example
of memory leakage.  Could it please be annotated with an unambiguous
"This is how not to do it"?  Or a fuller working example added, such as:

> local udpskt
> udpskt = net.createUDPSocket()
> if udpskt ~= nil
> then
>     udpskt:on("sent", function(s, c) s:close() end)
>     udpskt:send(UDPPort, broadcast, "Message")
> end

Thanks - Will

marcelstoer, 2018-03-27:
The FAQ should actually just link to the relevant documentation/example of the net module so that code doesn't have to be updated in two places. Looking forward to your PR.

joysfera, 2018-03-27:
I believe it indeed documents how easy it is to create a memory leak, and also an example of a programmer's bug, not a Lua/NodeMCU bug. Linking to the net module documentation is not appropriate there. The "This is how not to do it" would be enough. Looking forward to my PR :)

marcelstoer, 2018-03-27:
Sure, you can declare it as a bad example (good idea) but will also have to present how to do it properly. This can either be done by giving another example (that fixes the first one) or by linking to a good example in the respective module.

marcelstoer, 2018-03-31:
> Looking forward to my PR :)

Would be cool if we got this before the next master drop, thanks.

joysfera, 2018-03-31:
When is the next master drop going to happen?

marcelstoer, 2018-03-31:
Monday evening if nothing unexpected happens.

PR #2339 (2018-04-02): net socket documentation clarification
Hopefully helps to fix #2321.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.
- [ ] The code changes are reflected in the documentation at `docs/en/*`.

My first PR (ever), untested because I haven't managed to convince the RTD to build the docs, sorry.
closed 2018-04-02 by marcelstoer