nodemcu/nodemcu-firmware #2323: Unable to open tcp server on port 80 after enduser_setup

makefu, 2018-03-26:
### Expected behavior
after running `enduser_setup` it should be possible to open a new tcp server on port 80. 
### Actual behavior
Trying to open a new tcp listening socket on port 80 after enduser_setup results in `address in use`:

```lua
PANIC: unprotected error in call to Lua API (init.lua:5: address in use)
```
This still fails even if the creating of the listening port is deferred by 10 seconds. It should be at least possible to manually stop enduser_setup from blocking the tcp port.

### Test code

```Lua
enduser_setup.start(
  function ()
    local srv=net.createServer(net.TCP,30)
    srv:listen(80,function(conn)
        conn:on("receive", handle_request)
    end)
  end)
```

### NodeMCU version
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 477116f79f9e0274c02c6287ed20e637d7c32d56
	SSL: false
	modules: color_utils,cron,enduser_setup,file,gpio,http,mqtt,net,node,sjson,tmr,uart,wifi,ws2812,ws2812_effects
 build 	built on: 2018-03-26 20:01
 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
```

### Hardware
no extra hardware required

marcelstoer, 2018-03-27:
You can't create a new listener on 80 in that callback as enduser-setup still holds on to it as you found out:

-> https://nodemcu.readthedocs.io/en/master/en/modules/enduser-setup/#parameters_1

closed 2018-03-27 by marcelstoer

makefu, 2018-03-27:
Thanks for the heads-up, maybe a quick note could be added to the module documentation.
For future reference, this is the code i came up with:
```
enduser_setup.start(
  function ()
    tmr.alarm(1,1000,0, function()
      print("stopping enduser_setup")
      enduser_setup.stop()
      local srv=net.createServer(net.TCP,30)
      srv:listen(80,function(conn)
          conn:on("receive", print)
      end)
    end)
  end)
```