nodemcu/nodemcu-firmware #2325: building esp32 fresh, make errors "ctype.h: No such file or directory"

danlu01, 2018-03-27:
I'm trying the build nodemcu ESP32 on linux following this doc:
https://nodemcu.readthedocs.io/en/dev-esp32/en/build/

I installed Linux Mint 64bit in a Oracle VM.  Then ran:
git clone --branch dev-esp32 --recurse-submodules https://github.com/nodemcu/nodemcu-firmware.git nodemcu-firmware-esp32

When I try to run make, I got the following error:
daniel@daniel-VirtualBox ~/nodemcu-firmware-esp32 $ make
Setting IDF_PATH and re-invoking...
make[1]: Entering directory '/home/daniel/nodemcu-firmware-esp32'
make[2]: Entering directory '/home/daniel/nodemcu-firmware-esp32/sdk/esp32-esp-idf/tools/kconfig'
cc  -DCURSES_LOC="<curses.h>" -DLOCALE -MD   -c -o mconf.o mconf.c
mconf.c:11:19: fatal error: ctype.h: No such file or directory
compilation terminated.

What an I doing wrong?

devsaurus, 2018-03-31:
Can't reproduce this on `dev-esp32`.
I'll prepare an SDK update within the next days, maybe this has some positive side effects for your configuration.

devsaurus, 2018-04-04:
I've updated the SDK, but this error is most likely related to your build system.
Please double check that packages for glibc development are installed (e.g. `glibc-devel`).

devsaurus, 2018-04-15:
Closing, this is not a firmware issue.

closed 2018-04-15 by devsaurus