nodemcu/nodemcu-firmware #2326: Periodic Wifi disconnects with SDK 2.2.1

Code1110, 2018-03-27:
### Expected behavior
Node will not disconnect from Wifi.

### Actual behavior
Depending on the selected set of modules, the node disconnects roughly once per hour from Wifi.

NO disconnects with this build (SDK 2.1.0):
```
NodeMCU custom build by frightanic.com
	branch: master
	commit: 5073c199c01d4d7bbbcd0ae1f761ecc4687f7217
	SSL: false
	modules: adc,bit,file,gpio,http,i2c,mqtt,net,node,ow,rtctime,sjson,sntp,spi,tmr,uart,wifi
 build 	built on: 2018-03-27 16:43
 powered by Lua 5.1.4 on SDK 2.1.0(116b762)
```

NO disconnects with this build (SDK 2.2.1, different modules than with SDK 2.1.0):
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 477116f79f9e0274c02c6287ed20e637d7c32d56
	SSL: false
	modules: enduser_setup,file,gpio,http,i2c,mqtt,net,node,ow,rtctime,si7021,sjson,sntp,tmr,tsl2561,uart,wifi
 build 	built on: 2018-03-26 17:50
 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
```

Disconnects (SDK 2.2.1, same modules as with SDK 2.1.0):
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 477116f79f9e0274c02c6287ed20e637d7c32d56
	SSL: false
	modules: adc,bit,file,gpio,http,i2c,mqtt,net,node,ow,rtctime,sjson,sntp,spi,tmr,uart,wifi
 build 	built on: 2018-03-27 16:34
 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
```

Disconnect log:
```
=== Start Logging ===
Got Time: 27.03.2018 16:40:25
27.03.2018 17:14:59	STA - DISCONNECTED	SSID: [SSID]	BSSID: [BSSID]	reason: 200
27.03.2018 17:15:22	STA - DISCONNECTED	SSID: [SSID]	BSSID: [BSSID]	reason: 200
27.03.2018 17:33:36	STA - DISCONNECTED	SSID: [SSID]	BSSID: [BSSID]	reason: 200
27.03.2018 17:47:22	STA - DISCONNECTED	SSID: [SSID]	BSSID: [BSSID]	reason: 200
27.03.2018 17:52:21	STA - DISCONNECTED	SSID: [SSID]	BSSID: [BSSID]	reason: 200
27.03.2018 17:52:47	STA - DISCONNECTED	SSID: [SSID]	BSSID: [BSSID]	reason: 200
27.03.2018 17:53:10	STA - DISCONNECTED	SSID: [SSID]	BSSID: [BSSID]	reason: 200
27.03.2018 17:53:42	STA - DISCONNECTED	SSID: [SSID]	BSSID: [BSSID]	reason: 200
27.03.2018 20:19:48	STA - DISCONNECTED	SSID: [SSID]	BSSID: [BSSID]	reason: 200
```

I did the tests in parallel with two different nodes, that had different firmware installed. The test scripts were always the same.

### Test code
**Flash settings**
Flash tool: NodeMCU-PyFlasher-3.0-x64.exe
Baud rate: 921600
Flash mode: DIO
Erase flash: yes

**init.lua**
```Lua
wifi.setmode(wifi.STATION)

function logToFile(text)
	local f = file.open("wifievents.log", "a+")
	f:writeline(text)
	f:close()
end

station_cfg =
{
    ssid="ssid",
    pwd="pwd",
    save=false,
    auto=true,
    got_ip_cb=function(info) 
			print("Got IP: "..info.IP)
			logToFile("=== Start Logging ===")
			sntp.sync(nil, function () 
				print("Got Time")
				local tm = rtctime.epoch2cal(rtctime.get())
				local d = string.format("%02d.%02d.%04d %02d:%02d:%02d", tm.day, tm.mon, tm.year, tm.hour, tm.min, tm.sec)
				logToFile("Got Time: "..d)
				dofile("logWifiEvents.lua")
			end)
		end
}
wifi.sta.config(station_cfg)
```

**logWifiEvents.lua**
```Lua
function logEvent(event)
	print(event)
	local tm = rtctime.epoch2cal(rtctime.get())
	local d = string.format("%02d.%02d.%04d %02d:%02d:%02d", tm.day, tm.mon, tm.year, tm.hour, tm.min, tm.sec)
	logToFile(d..event)
end

wifi.eventmon.register(wifi.eventmon.STA_CONNECTED, function(T)
	logEvent("\tSTA - CONNECTED".."\tSSID: "..T.SSID.."\tBSSID: "..T.BSSID.."\tChannel: "..T.channel)
end)

wifi.eventmon.register(wifi.eventmon.STA_DISCONNECTED, function(T)
	logEvent("\tSTA - DISCONNECTED".."\tSSID: "..T.SSID.."\tBSSID: "..T.BSSID.."\treason: "..T.reason)
end)

wifi.eventmon.register(wifi.eventmon.STA_AUTHMODE_CHANGE, function(T)
	logEvent("\tSTA - AUTHMODE CHANGE".."\told_auth_mode: "..T.old_auth_mode.."\tnew_auth_mode: "..T.new_auth_mode)
end)

wifi.eventmon.register(wifi.eventmon.STA_GOT_IP, function(T)
	logEvent("\tSTA - GOT IP".."\tStation IP: "..T.IP.."\tSubnet mask: "..T.netmask.."\tGateway IP: "..T.gateway)
end)

wifi.eventmon.register(wifi.eventmon.STA_DHCP_TIMEOUT, function()
	logEvent("\tSTA - DHCP TIMEOUT")
end)

 wifi.eventmon.register(wifi.eventmon.WIFI_MODE_CHANGED, function(T)
	logEvent("\tSTA - WIFI MODE CHANGED".."\told_mode: ".. T.old_mode.."\tnew_mode: "..T.new_mode)
end)
```

### NodeMCU version
see above

### Hardware
WeMos D1 mini

dnc40085, 2018-03-28:
Hello, your disconnect reason (200) is a "beacon timeout", which means the ESP8266 has not received a beacon packet from the router/access point(AP) in a while, so it assumes that it has lost connection with the AP and tries reconnecting to the AP.

This could be caused by the ESP8266 not having a good signal(RSSI), RSSI can be checked with `wifi.sta.getrssi()`, the closer RSSI is to 0 the better. If you have a good signal while connected to the AP and are still getting disconnects, you can try power-cycling your AP or performing a factory reset and if either of those don't work I would suggest trying a different AP to see if the problem persists.

In my experience with most routers/APs, after a couple of weeks(or months) all wifi connections start seeing degraded performance and just get progressively worse over time. Sometimes power-cycling fixes the problem, other times the only way to fix it is to do a full factory reset on the AP. I'm pretty sure all this is the result of crappy firmware, but not all routers/APs are compatible with OpenWRT, so we are stuck with what we got.

devsaurus, 2018-03-28:
I can't reproduce yet, running a comparable build which disconnects for you
```
NodeMCU custom build by frightanic.com
	branch: dev
	commit: 477116f79f9e0274c02c6287ed20e637d7c32d56
	SSL: false
	modules: adc,bit,file,gpio,http,i2c,mqtt,net,node,ow,rtctime,sjson,sntp,spi,tmr,uart,wifi
 build 	built on: 2018-03-28 08:17
 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
```

Your code doesn't save the wifi configuration, maybe there's some bogus config still stored in flash. You could try to clear that with [`wifi.sta.clearconfig()`](https://nodemcu.readthedocs.io/en/dev/en/modules/wifi/#wifistaclearconfig) and check for periodic disconnects again.

Code1110, 2018-03-31:
@dnc40085 Thank you for your answer. The two nodes I'm using for testing are very close to the AP, so I don't think the signal is the problem. Besides this I have more than 10 nodes that are connected permanently to this AP and they don't suffer from the disconnect issue. But it is anyways a good idea to do the test with a different AP.

@devsaurus Thanks for testing. Yesterday I issued the `wifi.sta.clearconfig()` command on the node that had disconnects and they are gone so far. It could have something to do, that I flashed several different firmware versions from the dev branch on that node. I will do some more testing and report back.

Anyways the new firmware works fine for me and it solves a lot of issues.

devsaurus, 2018-04-01:
> But I'm wondering, why the wifi config was in flash, because I always erased the flash, while loading a new firmware.

I also don't have an explanation for this atm.

closed 2018-06-23 by marcelstoer