nodemcu/nodemcu-firmware #2327: Cleaning up the .build directory.

cwrseck, 2018-03-28:
I've been messing around with the integer build of luac.cross on the current
dev branch, which fails when it can't find c_strtol() (for various reasons
it isn't worth a developers time).  However, I keep forgetting to remove
the .build directory; "make clean" should certainly do this, and it should
probably be the first step in tools/cross-lua.lua

Will

TerryE, 2018-03-28:
Will, If you don't mind I will close this because the LFS patch (which I will be pushing into dev as soon as we have staged dev into master) has already moved the `Luac.cross` build into the standard make and builds integer builds fine.

We won't be fixing this for older versions.

closed 2018-03-28 by TerryE

cwrseck, 2018-03-31:
"make clean" is generally expected to remove intermediate build files.

Will

TerryE, 2018-03-31:
Yes.  So what? Don't you think that I don't know what `make clean` is supposed to do, for some strange reason?

I closed this, because I tagged it as **wontfix** because we've already got a wider change which fixes this anyway in the pipeline.

cwrseck, 2018-04-03:
Thanks - I didn't realise it was among the pending fixes.

Will

TerryE, 2018-04-03:
:wink: So you didn't bother to read my first reply then. :smile:

cwrseck, 2018-04-04:
Yes, I read your reply.  Your language was imprecise.

Will