nodemcu/nodemcu-firmware #2328: [net.TCP] Sometimes we receive concat messages

Lir10, 2018-03-29:
Testing with the latest firmware (master)

Looks like sometimes when we send JSON messages from the server to the ESP client they are not seperated for example:

python server sends the following JSONS one by one:
```

'{"test":1}\r\n'
'{"test":2}\r\n'
```

ESP Client code for receiving:

```
conn:on("receive",function(t,e)
  print("Received ", e)
end)
```

Results:
`Received '{"test":1}\r\n{"test":2}\r\n'`

Any ideas why ESP concat those messages?

pjsg, 2018-03-29:
This is a characteristic of TCP. TCP does not have record boundaries and hence you should expect to see any TCP data delivered in one or more callbacks.

closed 2018-03-29 by pjsg

Lir10, 2018-03-29:
@pjsg Thanks for the response. 
Any ideas how we should handle this to exract the JSON from it ? do we need to split the string on every packet we receive?