nodemcu/nodemcu-firmware #2329: Out of memory when conn:send fails multiple times

Lir10, 2018-03-29:
Looks like i's a bug but when we `conn:send` fails it just crashes with [out of memory]

Maybe it's possible to clean the TCP buffer when multiple messages are not sent?

TerryE, 2018-03-30:
@Lir10 GC does work.  If its not collecting something, then that's because your got dangling references.  See my FAQ discussions.

Again re multiple calls to send, read my FAQ.  This is breaking the rule.  The SDK in effect does non-blocking I/O and multiple sends per task are not supported.   You need to space then out with on("sent") callbacks.

So tIMO the only bug here is that you have'nt read and followed the documentation :smile:

closed 2018-03-30 by TerryE

Lir10, 2018-03-30:
Hi @TerryE Thanks for the response. 

Actually i've read the documentation but couldn't understand what we should do if on("send") callback doesn't get called. 

For example i'm sending a packet but it didn't delivered - should i continue to send the next packets event until i get "out of memory" ?

TerryE, 2018-03-30:
TCP is an assured ordered communication so packets will get sent unless there is a link error and in which case  you will receive a reconnection event.  OK, this is a bit of a misnomer, butt he API documentation does clarify that this is used for abnormal disconnections.  
Also the non-OS makes it quite clear that you can ony issue one send per task execution, so you nedd to break up your code using on events to comply with this (Or use `node.task.post()` for each  subsequent one, but just remember that network I/O is non-blocking.  if you don't flow limit these sends then the non-OS SDK will reboot the ESP module.

Lir10, 2018-04-01:
Hi @TerryE  Maybe you could provide an example on how we should use this in the right way? `node.task.post()` is not mentioned in the `net ` module docs

TerryE, 2018-04-01:
> See my FAQ

Lir10, 2018-04-01:
Thanks @TerryE appreciate your help!