nodemcu/nodemcu-firmware #2331: Documentation: displayed width

dnc40085, 2018-03-30:
### Missing feature
It would be nice if the NodeMCU documentation filled most of the available screen space on [nodemcu.readthedocs.io](nodemcu.readthedocs.io).

### Justification
Readability of documentation and most monitors have a aspect ratio of 16:9 these days.

### Workarounds
Add `div.wy-nav-content{ max-width:98% }` to `docs/css/extra.css` and build documentation locally

If others are in favor of this change, I can submit a PR with the necessary changes.

Side note: Thank you @marcelstoer for the improved documentation, the github wiki was a pain.

marcelstoer, 2018-03-30:
This is a tricky one but I vote reject this.

I'm not a UX expert but from personal experience I know that long lines significantly reduce legibility of a text. So, I quickly searched for UX guidelines and found, no surprise, different recommendations but none advises to use more than 90 characters per line. Sources: https://ux.stackexchange.com/a/3623,  https://baymard.com/blog/line-length-readability.
The RTD theme limits the "content width" to 800px. If we increased it to 1100px say we would end up with roughly 135 characters per line. It would be really hard to read.

Also, as we're using a commonly used theme I'd argue that we most likely serve the average reader well.

dnc40085, 2018-04-01:
>https://baymard.com/blog/line-length-readability

I guess I can't argue with 28,000+ hours of UX research that I didn't know existed in the first place, didn't even know UX guidelines were a thing. Thank you for your response.

closed 2018-04-01 by dnc40085