nodemcu/nodemcu-firmware #2336: DHCP and sethostname() duplicated calls (fixable?)

cwrseck, 2018-03-31:
With the 2.1.0 20170824 master firmware, running the code

>     wifi.eventmon.register(wifi.eventmon.STA_GOT_IP,
>         function(T) print("STA_GOT_IP") end)
> 
>     -- Set up the wireless connection.
>     wifi.setmode(wifi.STATION)
>     -- wifi.sta.sethostname("greenhouse")
>     wifi.setphymode(wifi.PHYMODE_G)
> 
>     -- Use the new getap() output format.
>     wifi.sta.getap(1,
>        function(t)
>             local sta_cfg = { }
>             sta_cfg.ssid = AP
>             sta_cfg.pwd = APS[AP]
>             sta_cfg.auto = true
>             wifi.sta.config(sta_cfg)
>             end
>         end)

gives the following dhcpd.log: on the Access Point:

> 19:43:45 farthing dhcpd: DHCPDISCOVER from 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0
> 19:43:46 farthing dhcpd: DHCPOFFER on 193.237.32.226 to 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0
> 19:43:46 farthing dhcpd: DHCPREQUEST for 193.237.32.226 (193.237.32.225) from 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0
> 19:43:46 farthing dhcpd: DHCPACK on 193.237.32.226 to 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0
> 19:43:46 farthing dhcpd: DHCPREQUEST for 193.237.32.226 (193.237.32.225) from 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0
> 19:43:46 farthing dhcpd: DHCPACK on 193.237.32.226 to 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0
> 19:47:37 farthing dhcpd: DHCPREQUEST for 193.237.32.199 from 00:1c:25:1e:39:74 via eth0
> 19:47:37 farthing dhcpd: DHCPACK on 193.237.32.199 to 00:1c:25:1e:39:74 via eth0
> 19:48:33 farthing dhcpd: DHCPDISCOVER from 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0
> 19:48:34 farthing dhcpd: DHCPOFFER on 193.237.32.226 to 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0
> 19:48:35 farthing dhcpd: DHCPREQUEST for 193.237.32.226 (193.237.32.225) from 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0
> 19:48:35 farthing dhcpd: DHCPACK on 193.237.32.226 to 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0
> 19:48:35 farthing dhcpd: DHCPREQUEST for 193.237.32.226 (193.237.32.225) from 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0
> 19:48:35 farthing dhcpd: DHCPACK on 193.237.32.226 to 5c:cf:7f:15:34:e6 (NODE-1534E6) via wlan0

The doubled ACK/REQUESTS are almost certainly bugs - I'm not sure about the
repeated DISCOVER.  If the sethostname() call is uncommented things are even
more confused, with calls for both the default and the new hostnames.  The IP
packets, however, are still correctly transmitted.

I _think_ this problem lies deep in the Espressif firmware, and isn't fixable;
though a working sethostname() would be nice.  Does anyone know different?

Will

devsaurus, 2018-03-31:
You're using [wifi.sta.config()](https://nodemcu.readthedocs.io/en/master/en/modules/wifi/#wifistaconfig) with `auto=true` and implied `save=true`. I expect 2 consequences out of this:
1. Your module connects *automatically* after reset with the *saved* configuration.
  This would generate the first DHCP cycle.
  IIRC the hostname provided with `wifi.sta.sethostname()` is not saved in flash.
2. Your script then re-connects wifi actively due the call to `wifi.sta.config()`.

To have a controlled environment for further analysis, I recommend to
* erase configuration history with [`wifi.sta.clearconfig()`](https://nodemcu.readthedocs.io/en/master/en/modules/wifi/#wifistaclearconfig) once
* add `save=false` to the `sta_cfg` table

Does the server log change?

cwrseck, 2018-04-03:
With the 2.1.0 20170824 master firmware, running the revised code below:

>     -- Disable autoconnect.
>     wifi.sta.autoconnect(0)
>     wifi.eventmon.register(wifi.eventmon.STA_GOT_IP,
>         function(T) print("STA_GOT_IP") end)
> 
>     -- Set up the wireless connection.
>     wifi.setmode(wifi.STATION)
>     wifi.setphymode(wifi.PHYMODE_G)
>     wifi.sta.clearconfig()
>     -- Default hostname is NODE-1534E6
>     wifi.sta.sethostname("greenhouse")
> 
>     -- The auto argument avoids need for explicit sta.connect()
>     -- This isn't the SDK auto_connect flag.
>     wifi.sta.getap(1,
>         function(t)
>             local sta_cfg = { }
>             sta_cfg.ssid = SSID
>             sta_cfg.pwd = pwd
>             sta_cfg.auto = true
>             sta_cfg.save = false
>             wifi.sta.config(sta_cfg)
>             -- Needed only if auto is false
>             -- wifi.sta.connect()
>             end
>         end)

gives (with a 120 second lease time):

> 17:18:34 farthing dhcpd: DHCPDISCOVER from 5c:cf:7f:15:34:e6 (greenhouse) via wlan0
> 17:18:35 farthing dhcpd: DHCPOFFER on 193.237.32.226 to 5c:cf:7f:15:34:e6 (greenhouse) via wlan0
> 17:18:35 farthing dhcpd: DHCPREQUEST for 193.237.32.226 (193.237.32.225) from 5c:cf:7f:15:34:e6 (greenhouse) via wlan0
> 17:18:35 farthing dhcpd: DHCPACK on 193.237.32.226 to 5c:cf:7f:15:34:e6 (greenhouse) via wlan0
> 17:18:35 farthing dhcpd: DHCPREQUEST for 193.237.32.226 (193.237.32.225) from 5c:cf:7f:15:34:e6 (greenhouse) via wlan0
> 17:18:35 farthing dhcpd: DHCPACK on 193.237.32.226 to 5c:cf:7f:15:34:e6 (greenhouse) via wlan0
> 17:19:22 farthing dhcpd: DHCPREQUEST for 193.237.32.226 (193.237.32.225) from 5c:cf:7f:15:34:e6 (greenhouse) via wlan0
> 17:19:22 farthing dhcpd: DHCPACK on 193.237.32.226 to 5c:cf:7f:15:34:e6 (greenhouse) via wlan0
> 17:20:22 farthing dhcpd: DHCPREQUEST for 193.237.32.226 (193.237.32.225) from 5c:cf:7f:15:34:e6 (greenhouse) via wlan0
> 17:20:22 farthing dhcpd: DHCPACK on 193.237.32.226 to 5c:cf:7f:15:34:e6 (greenhouse) via wlan0

The initial REQUEST / ACK pair is still duplicated, but the DHCPDISCOVER call
isn't.  I suspect that the DHCP state machine is being started at the wrong
node; it's a bug, but scarcely fatal.  The good news is that now that the
configuration isn't being saved, the duplicate NODE-1534E6 / greenhouse calls
have gone.  Many thanks for that advice.

Will

devsaurus, 2018-04-03:
> The good news is that now that the
configuration isn't being saved, the duplicate NODE-1534E6 / greenhouse calls
have gone.

Indeed, we're closer to isolate the suspected bug.

> The initial REQUEST / ACK pair is still duplicated

Could this be an artifact of the short lease time (120&nbsp;s)? I'm asking because the dhcp coarse timer period is 60&nbsp;s and integer rounding might end up with a marginal duration when your lease renewal time is close to the timer's period. And I don't see the duplicated REQUEST / ACK sequence here with a lease time of 432000&nbsp;s.

cwrseck, 2018-04-05:
The duplicate Request/Acknowledge transmissions seem to be a mare's nest
full of red herrings; looking at the packet exchange with Wireshark there
are two initial Discover packets with the same transaction ID, then a single
Offer followed by two Requests/Acknowledge.

However, an ordinary (laptop) wireless connection shows the same sequence,
so it has to be a problem with the local AP software.  I'll check again with
another access point, but I'm now pretty sure it has nothing to do with
NodeMCU.

Will

TerryE, 2018-04-11:
>  but I'm now pretty sure it has nothing to do with NodeMCU.

closed 2018-04-11 by TerryE