nodemcu/nodemcu-firmware #2338: MQTT over TLS with ECC

joysfera, 2018-04-01:
This may not be a real bug report (yet) but I've just failed to test my idea how to save precious memory with TLS connection and I don't know if it is my fault or NodeMCU's. In #1699 I found out that MQTT over secure connection takes about 17 kB of memory on the heap. The idea is to try out installing ECC certificate on the MQTT server to see if it saves memory on the NodeMCU client as well.

ECC https://wiki.openssl.org/index.php/Elliptic_Curve_Cryptography uses 6-30 times shorter keys (e.g. 224 bytes instead of 2 kB) so there's hope to save memory, right?

### Expected behavior
NodeMCU MQTT connects and proceeds normally.

### Actual behavior
MQTT fails to connect with reason = -5

### Test code
```Lua
m=mqtt.Client(node.chipid(), 120)
m:connect("eccbroker", 8883, 1, 0, function(c) print("OK") end, function(c, r) print(r) end)
```
mosquitto_pub connects OK so the broker is configured well.

NodeMCU used to connect to my broker OK until I replaced the SSL certificates with ECC ones (generated by LE using ACME.sh).

### NodeMCU version
Lua 5.1.4 on SDK 2.2.1(cfd48f3)
branch: dev  commit: 9af4f58417103a21d31f70fc6e44bfba55d36789

### Hardware
plain ESP8266 v07

marcelstoer, 2018-04-01:
There are some cipher suite configs in `app/include/user_mbedtls.h` mentioned in the docs right below that sentence and the full config in `app/include/mbedtls/config.h`.

TerryE, 2018-04-01:
I've just made this point on another issue, but I've really got to ask myself:  we we expect LUA application developers to implement LTS or MQTT over TLS without it?

If the answer is no, then wat some point in the near future, we should just embrace the fact that you can't run TLS without LFS enables and exploit LFS to simplify the implementation.

nwf, 2019-07-21:
ECDSA ciphers are now enabled in TLS and I have successfully brought up MQTT connections using them, but only with LFS; even my modest application running in RAM was enough to deplete memory too much for TLS negotiation.

closed 2019-07-21 by nwf