nodemcu/nodemcu-firmware #2341: DS18B20 eats up memory over time

StefanL38, 2018-04-03:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
uses a constant amount of memory

### Actual behavior
repetitive calls of ds18b20.read( function(ind,rom,res,temp,tdec,par) ...) 
eats up memory over time slowly. After about 350 calls error E:M xxxx out of memory
### Test code
```Lua
tmr.alarm(tmr_DS18B20_reading, tmr_Sensor_interval, 1, function ()

  ds18b20.read(
    function(ind,rom,res,temp,tdec,par)
      
      if rom == FeZi_ROM_ID then
        FeZi_Temp = temp
        FeZi_SollTemp = FeZi_NachtTemp -- Set lower night-temperature as default
        if (MinuteOfDay > FeZi_TagMinAnf) and (MinuteOfDay < FeZi_TagMinEnd) then
          FeZi_SollTemp = FeZi_TagTemp -- set day-temperature 
        end  

        if FeZi_Temp < FeZi_SollTemp then
          gpio.write(FeZi_Pin, gpio.HIGH)
          DebugOut("FeZi_Pin, gpio.HIGH",PrintDebugOutPut)
        end 

        if FeZi_Temp > (FeZi_SollTemp + OnOff_Hysteresis) then
          gpio.write(FeZi_Pin, gpio.LOW)
          DebugOut("FeZi_Pin, gpio.LOW",PrintDebugOutPut)
        end 
      end --if rom == FeZi_ROM_ID then  

      print(node.heap())
      print(" FeZi:"..FeZi_Temp..")
    end,{}); --ds18b20.read(function(ind,rom,res,temp,tdec,par)

``` I don't post the full code here as I estimate that the reason for eating up the memory is caused by bad
readings due to a 1wire-network that works on the edge of the hardware-specs so that only some readings fail. Anyway I tested the exact same 1wire-network with the ow-module and this does NOT eat up memory over time.  

### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
master and dev

### Hardware

WeMos D1 mini a 1wire-network with 6 DS18B20 sensors bus-topology star radius 20m, weight 60-80m
each branch with a 140 Ohm-resistor. Each sensor supplied with 5V
8<------------------------ END BUG REPORT -------------------------------------

peturdainn, 2018-04-03:
would be interesting to also post that timer interval, and some POC that really only does the reads?

FWIW I'm busy setting up several NodeMCU with DS18b20 (but only one per module), and with a read interval of 5 seconds (or >100K reads) they have all been running for many days without restarts (they report to a logging server when starting or in case of errors).

So either related to reading multiple DS18b20, or something else in the code

cwrseck, 2018-04-03:
Using the 2.1.0 20170824 master firmware to read three DS18B20s shows no memory
leaks after two months running, with a 6 minute read interval.

Will

TerryE, 2018-04-03:
Incidentally, you only need to do the closure once not once per sec:
```Lua
local function ds18b20_cb((ind,rom,res,temp,tdec,par)
-- body goes here
end
local function tmr_DS18B20_cb()
  ds18b20.read(ds18b20_cb)
end
tmr.alarm(tmr_DS18B20_reading, tmr_Sensor_interval, 
          tmr.ALARM_AUTO,      tmr_DS18B20_cb)
```

closed 2018-04-03 by TerryE