nodemcu/nodemcu-firmware #2346: Performance regression with http module on 2.2

heythisisnate, 2018-04-05:
### Expected behavior
Response time of simple http(s) requests should be reasonably fast and comparable to previous NodeMCU firmware versions.

### Actual behavior
On the latest 2.2 firmware, a simple https GET is ~5x slower than on NodeMCU 1.5.4.1

### Test code
```Lua
busy = false
tmr.create():alarm(1000, tmr.ALARM_AUTO, function()
  if not busy then
    local start_time = tmr.now()
    busy = true
    http.get('https://api.ipify.org', nil, function(code, data)
      print(code, "Elapsed time:", (tmr.now() - start_time)/1000 .. 'ms')
      busy = false
    end)
  end
end)
```
The above code will repeatedly do a simple GET to a trivial public endpoint and print out the time that it takes to do the call and execute the callback. The response data is not relevant in this case.

When run on master (2.2.0):
```
200	Elapsed time:	1858ms
200	Elapsed time:	1840ms
200	Elapsed time:	1855ms
200	Elapsed time:	1876ms
200	Elapsed time:	1860ms
200	Elapsed time:	1880ms
200	Elapsed time:	1869ms
200	Elapsed time:	1863ms
200	Elapsed time:	1950ms
200	Elapsed time:	1890ms
200	Elapsed time:	1951ms
200	Elapsed time:	1883ms
200	Elapsed time:	1900ms
200	Elapsed time:	1864ms
200	Elapsed time:	1869ms
200	Elapsed time:	1848ms
200	Elapsed time:	1889ms
200	Elapsed time:	1879ms
200	Elapsed time:	1869ms
200	Elapsed time:	1880ms
```

When run on 1.5.4.1-final:
```
200	Elapsed time:	355ms
200	Elapsed time:	338ms
200	Elapsed time:	335ms
200	Elapsed time:	348ms
200	Elapsed time:	340ms
200	Elapsed time:	331ms
200	Elapsed time:	337ms
200	Elapsed time:	336ms
200	Elapsed time:	358ms
200	Elapsed time:	392ms
200	Elapsed time:	350ms
200	Elapsed time:	333ms
200	Elapsed time:	336ms
200	Elapsed time:	338ms
200	Elapsed time:	337ms
200	Elapsed time:	344ms
200	Elapsed time:	335ms
200	Elapsed time:	329ms
200	Elapsed time:	330ms
200	Elapsed time:	330ms
```

As you can see, when run on 1.5.4.1, the request completes about 5 times faster than on the newest firmware. Is this expected? What could be attributed to the slowdown? I've tested with various public API endpoints with similar results. I'm happy to try to look into this problem myself, but wondering if anyone has some insight or pointers on where to look.

### NodeMCU version
2.20-master (https://github.com/nodemcu/nodemcu-firmware/commit/67027c0d05f7e8d1b97104e05a3715f6ebc8d07f)

### Hardware
NodeMCU dev board (LoLin version)

marcelstoer, 2018-04-05:
Did you try running this against plain-HTTP endpoints as well (no TLS)?

marcelstoer, 2018-04-05:
> the performance issue is related to TLS

Nearly 2s for a single request is terrible - particularly for battery powered devices. As we've shown in [a recent blog post](https://thingpulse.com/2018/02/25/esp8266-monitoring-power-consumption/) it's the energy consumption during the WiFi-on phase that almost exclusively determines the total consumption over time. Deep-sleep consumption is rather negligible. So, every 100ms you waste really hurts.

@heythisisnate you started this with the statement/subject "Performance regression with http". Does this imply the TLS module is not affected?

@nwf do you have any idea, anything that looked odd during the mbedTLS upgrade?

heythisisnate, 2018-04-06:
digging deeper, I'm pretty sure the slowness is coming from `mbedtls_ssl_handshake`. I'm having trouble figuring out how to turn on mbedTLS debugging output. Anyone know how?

marcelstoer, 2018-04-06:
I haven't checked but maybe `MBEDTLS_SSL_DEBUG_ALL` in `app/include/mbedtls/config.h` is what you're looking for.

devsaurus, 2018-04-06:
> Anyone know how?

My best guess would be the following settings (can't test atm):
1. Define `MBEDTLS_SSL_DEBUG_C` and maybe `MBEDTLS_SSL_DEBUG_ALL` in [`user_mbedtls.h`](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/include/user_mbedtls.h#L133)
2. Set the debug threshold with `tls.setDebug(3)`. Ref [`debug.h`](https://github.com/nodemcu/nodemcu-firmware/blob/dev/app/include/mbedtls/debug.h#L83) for threshold values.

marcelstoer, 2018-04-06:
> > Does this imply the TLS module is not affected?
> 
> No, please do not assume that. 

I adjusted the issue title accordingly.

nwf, 2018-04-06:
I didn't see anything that looked alarming, but I recall that there is some clock fiddling around the SSL handshake.  I wonder if somehow the CPU clock is getting set to a smaller value rather than a faster one?

heythisisnate, 2018-04-06:
@nwf I noticed the CPU clock frequency fiddling too. It looks like `system_update_cpu_freq` is called before and after the handshake. I added some debugging and found that the value is _always_ set to 160 (at least in this test scenario). I never saw it changed to a different value. That doesn't make too much sense to me. Do you have any insight here?

nwf, 2018-04-06:
@heythisisnate I am not sure.  It's possible that some other environments down-clock the CPU, in which case it'd make sense to boost it around the SSL handshake?  Still, if it's always 160, that's probably not the problem. :(

heythisisnate, 2018-04-06:
I'm still having a tough time enabling mbedTLS debug output. Any help in this regard would be appreciated. I've tried setting the debug threshold, but when I do I get an unusable firmware that just constantly reboots with this:
```
_flash_used_end:402a0600
fs.start:b0000,max:34b000
_flash_used_end:402a0600
fs.start:b0000,max:34a000
mount res: 0, 0
Task task_lua started.
Fatal exception 0(IllegalInstructionCause):
epc1=0x2c223a25, epc2=0x00000000, epc3=0x00000000, excvaddr=0x402936da, depc=0x00000000
�
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```

I'm probably doing something wrong ... not too experienced with C.

nwf, 2018-04-06:
That's... interesting.  Unfortunately, I don't know what to do about it.  You might try just hard-coding the debug threshold rather than using `tls.setDebug` ?

devsaurus, 2018-04-06:
> Am I missing something?

Strange, this setup works for me on both integer and float builds
```
_flash_used_end:4029e070
fs.start:a0000,max:35b000
_flash_used_end:4029e070
fs.start:a0000,max:35a000
_flash_used_end:4029e070
fs.start:b0000,max:34b000
_flash_used_end:4029e070
fs.start:b0000,max:34a000
mount res: 0, 0
Task task_lua started.
nul mode, fpm auto sleep set:enalbe

NodeMCU 2.2.0.0 build unspecified powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
lua: cannot open init.lua
> Heap size::43112.
```
Looks like an inconsistent image - Did you flash both bin files?

heythisisnate, 2018-04-06:
> Strange, this setup works for me on both integer and float builds

Doh! this was totally my fault. I had written a little script that naïvely hardcoded some values. Didn't notice that the firmware with debugging was a bit larger. Now I'm getting plenty of debug output. Trying to make some sense of it.

heythisisnate, 2018-04-06:
Ok, here's the output from mbedTLS debug during the suspected slow part of the handshake:

```
TLS<2> (heap=6688): ssl_tls.c:6647 => handshake
TLS<2> (heap=6688): ssl_cli.c:3321 client state: 12
TLS<2> (heap=6688): ssl_tls.c:2442 => flush output
TLS<2> (heap=6688): ssl_tls.c:2454 <= flush output
TLS<2> (heap=6688): ssl_tls.c:4765 => parse change cipher spec
TLS<2> (heap=6688): ssl_tls.c:3763 => read record
TLS<2> (heap=6688): ssl_tls.c:2234 => fetch input
TLS<2> (heap=6688): ssl_tls.c:2392 in_left: 0, nb_want: 5
TLS<2> (heap=6688): ssl_tls.c:2416 in_left: 0, nb_want: 5
TLS<2> (heap=6688): ssl_tls.c:2417 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
TLS<2> (heap=6688): ssl_tls.c:2429 <= fetch input
TLS<3> (heap=6688): ssl_tls.c:3522 input record: msgtype = 20, version = [3:3], msglen = 1
TLS<2> (heap=6688): ssl_tls.c:2234 => fetch input
TLS<2> (heap=6688): ssl_tls.c:2392 in_left: 5, nb_want: 6
TLS<2> (heap=6688): ssl_tls.c:2416 in_left: 5, nb_want: 6
TLS<2> (heap=6688): ssl_tls.c:2417 ssl->f_recv(_timeout)() returned 1 (-0xffffffff)
TLS<2> (heap=6688): ssl_tls.c:2429 <= fetch input
TLS<2> (heap=6688): ssl_tls.c:3796 <= read record
TLS<3> (heap=6688): ssl_tls.c:4793 switching to new transform spec for inbound data
TLS<2> (heap=6688): ssl_tls.c:4843 <= parse change cipher spec
TLS<2> (heap=6688): ssl_cli.c:3321 client state: 13
TLS<2> (heap=6688): ssl_tls.c:2442 => flush output
TLS<2> (heap=6688): ssl_tls.c:2454 <= flush output
TLS<2> (heap=6688): ssl_tls.c:5401 => parse finished
TLS<2> (heap=6688): ssl_tls.c:5100 => calc  finished tls sha256
TLS<3> (heap=6688): ssl_tls.c:5124 dumping 'calc finished result' (12 bytes)
TLS<3> (heap=6688): ssl_tls.c:5124 0000:  d2 17 5c eb 2d 0b 27 d1 23 19 be d3              ..\.-.'.#...
TLS<2> (heap=6688): ssl_tls.c:5130 <= calc  finished
TLS<2> (heap=6688): ssl_tls.c:3763 => read record
TLS<2> (heap=6688): ssl_tls.c:2234 => fetch input
TLS<2> (heap=6688): ssl_tls.c:2392 in_left: 0, nb_want: 5
TLS<2> (heap=6688): ssl_tls.c:2416 in_left: 0, nb_want: 5
TLS<2> (heap=6688): ssl_tls.c:2417 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
TLS<2> (heap=6688): ssl_tls.c:2429 <= fetch input
TLS<3> (heap=6688): ssl_tls.c:3522 input record: msgtype = 22, version = [3:3], msglen = 40
TLS<2> (heap=6688): ssl_tls.c:2234 => fetch input
TLS<2> (heap=6688): ssl_tls.c:2392 in_left: 5, nb_want: 45
TLS<2> (heap=6688): ssl_tls.c:2416 in_left: 5, nb_want: 45
TLS<2> (heap=6688): ssl_tls.c:2417 ssl->f_recv(_timeout)() returned 40 (-0xffffffd8)
TLS<2> (heap=6688): ssl_tls.c:2429 <= fetch input
TLS<2> (heap=6688): ssl_tls.c:1604 => decrypt buf
TLS<2> (heap=6688): ssl_tls.c:2077 <= decrypt buf
TLS<3> (heap=6688): ssl_tls.c:3124 handshake message: msglen = 16, type = 20, hslen = 16
TLS<2> (heap=6688): ssl_tls.c:3796 <= read record
TLS<2> (heap=6688): ssl_tls.c:5469 <= parse finished
TLS<2> (heap=6688): ssl_cli.c:3321 client state: 14
TLS<2> (heap=6688): ssl_tls.c:2442 => flush output
TLS<2> (heap=6688): ssl_tls.c:2454 <= flush output
TLS<2> (heap=6688): ssl_cli.c:3432 handshake: done
TLS<2> (heap=6688): ssl_cli.c:3321 client state: 15
TLS<2> (heap=6688): ssl_tls.c:2442 => flush output
TLS<2> (heap=6688): ssl_tls.c:2454 <= flush output
TLS<3> (heap=6688): ssl_tls.c:5213 => handshake wrapup
TLS<3> (heap=6688): ssl_tls.c:5186 => handshake wrapup: final free
TLS<3> (heap=8712): ssl_tls.c:5206 <= handshake wrapup: final free
TLS<3> (heap=8712): ssl_tls.c:5268 <= handshake wrapup
TLS<2> (heap=8712): ssl_tls.c:6657 <= handshake
```

Unfortunately this doesn't mean much to me, and nothing obviously jumped out as slow as I was watching it output.

joysfera, 2018-04-06:
timestamps next to the heap size could be helpful for debugging slowness..

devsaurus, 2018-04-07:
Just a random thought... You're comparing firmware 1.5.4.1 (incl. axTLS) with 2.2.0 incl. mbedtls. The latter has a much larger feature set and supports TLS 1.2, whereas the baseline with axTLS stayed at TLS 1.1
Could this be the reason for the regression? Is it possible to restrict mbedtls to a comparable feature set like axTLS and run the performance test again? Might help to isolate the problematic aspect.

nwf, 2018-04-07:
@heythisisnate That all looks like a pretty healthy TLS bringup.  The only thing I see of note is that you appear to be negotiating TLS 1.2 (`version = [3:3]`) which, as @devsaurus notes, is not supported by the earlier axTLS stack.  It should be possible to nerf TLS 1.2 support by `#undef MBEDTLS_SSL_PROTO_TLS1_2`, but I've never tried.  It would seem odd to me if that's the problem.

I second @joysfera's suggestion, as I have no intuition about which part of that exchange is likely to be taking time.

FrankX0, 2018-04-10:
@joysfera 
> timestamps next to the heap size could be helpful for debugging slowness..

Change the following in espconn_mbedtls.c:
```
mbedtls_dbg(void *p, int level, const char *file, int line, const char *str)
{
	os_printf("TLS<%d> (heap=%d)(time=%d): %s:%d %s", level, system_get_free_heap_size(),system_get_time(), file, line, str);
}
```

vshymanskyy, 2018-09-24:
I'm now trying to run the test code for this issue on a recent build:
```log
NodeMCU custom build by frightanic.com
        branch: master
        commit: 3661b8d5eb5b42ed2d5ff51fa8e9628c17270973
        SSL: true
        modules: encoder,file,gpio,http,i2c,mdns,mqtt,net,node,rtcmem,rtctime,sntp,spi,tmr,uart,websocket,wifi,tls
 build created on 2018-09-20 20:55
 powered by Lua 5.1.4 on SDK 2.2.1(6ab97e9)
```
And I consistently get an error:
```log
HTTP client: Disconnected with error: -114
HTTP client: Connection timeout
-1      Elapsed time:   312.991ms
```
Is this also happening to anyone else?

closed 2019-10-03 by stale[bot]