nodemcu/nodemcu-firmware #2348: rotary module : channels 1 and 2 don't work

FredThx, 2018-04-07:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

The rotary module supports three switches. The channel is either 0, 1 or 2.

### Actual behavior

Only channel 0 work.

### Test code
```Lua
rotary.setup(0,1,2)
rotary.on(0,rotary.TURN, function(type, pos, when)
                print("0",type,pos,when)
            end)
rotary.setup(1,3,4)
rotary.on(1,rotary.TURN, function(type, pos, when)
                print("1",type,pos,when)
            end)
rotary.setup(2,5,6)
rotary.on(2,rotary.TURN, function(type, pos, when)
                print("2",type,pos,when)
            end)
```
### NodeMCU version
NodeMCU custom build by frightanic.com
	branch: master
	commit: 67027c0d05f7e8d1b97104e05a3715f6ebc8d07f
	SSL: false
	modules: bit,file,gpio,i2c,mqtt,net,node,rotary,sjson,tmr,u8g,uart,wifi
 build created on 2018-04-07 08:26
 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)

### Hardware
Wemos D1

8<------------------------ END BUG REPORT -------------------------------------

FredThx, 2018-04-07:
More information :
When I run the code above (test code) :

when I turn the rotary encoder 1 or 2 : no print

when turn the rotary encoder 0 :
The 3 callback are executed : 

0	8	17	40426626
0	8	18	40443874
1	8	17	40443884
2	8	18	40443889
0	8	19	40446912
1	8	18	40446937
2	8	19	40446944
0	8	20	40452503
0	8	21	40526655
1	8	20	40526662
2	8	21	40526671
0	8	20	40526698
1	8	21	40526704

pjsg, 2018-04-07:
I'll take a look

FredThx, 2018-04-09:
A lua solution : 
https://github.com/FredThx/nodemcu_iot/blob/master/rotary.lua

PR #2353 (2018-04-13): Missed setting the ID into the data block
Fixes #2348.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.

It turned out that there was an uninitialized filed in one of the data structures that was supposed to hold the channel number. It was always being set to zero and so channel zero would only ever work. Setting the field correctly, makes channels 1 and 2  [... cut]
closed 2018-04-13 by TerryE