nodemcu/nodemcu-firmware #2349: already connected error

shinji2009, 2018-04-08:
hi. sorry i am not a programmer so i am not quite understand http://nodemcu.readthedocs.io/en/dev/en/support/.

### Expected behavior
i was using this code for a long time and everuthing was fine after i google a lot and force it work because projects was old and a lot was changed in lua since then. it is a mix of two projects http://www.simpleiothings.com/build-a-10-wifi-smart-button/ and https://github.com/vChrisR/TheMagicButton 
### Actual behavior
now i want to move to nodemcu 2.2.0 because of krack and i got PANIC: unprotected error in call to Lua API (ifttt.lua:66: already connected). there is conn:connect(80,ip) at line 66
### Test code
```Lua
--ifttt.lua

--wifi.sta.disconnect()
  --wifi.sta.connect()

file.open( "iftttKey.lua", "r" )
iftttKey = file.read()
iftttKey = string.gsub(iftttKey, "%s+", "")
file.close()
--print("IFTTT Maker Key: "..iftttKey)

file.open( "eventName.lua", "r" )
eventName = file.read()
eventName = string.gsub(eventName, "%s+", "")
file.close()
print("IFTTT Event Name: "..eventName)

file.open( "sensorType.lua", "r" )
sensorType = file.read()
sensorType = string.gsub(sensorType, "%s+", "")
file.close()
print("Sensor Type: "..sensorType)

file.open( "sensorLocOne.lua", "r" )
sensorLocOne = file.read()
sensorLocOne = string.gsub(sensorLocOne, "%s+", "")
file.close()
print("Sensor Location: "..sensorLocOne)

file.open( "sensorLocTwo.lua", "r" )
sensorLocTwo = file.read()
sensorLocTwo = string.gsub(sensorLocTwo, "%s+", "")
file.close()
print("Sensor Location: "..sensorLocTwo)

conn = nil
conn=net.createConnection(net.TCP, 0) 
--net.createConnection(type, secure); i.e. Type = TCP,

conn:on("receive", function(conn, payload) 
    --Shutdown!
    gpio.write(4, gpio.LOW)
    tmr.alarm(0, 3000, 1, function()

        delay()

        end)
    end)

conn:on("connection", function(conn, payload) 

print("Sending to IFTTT")
conn:send("POST /trigger/" .. eventName .. "/with/key/" .. iftttKey .. "/?value1=" .. sensorType .. "&value2=" .. sensorLocOne .. "&value3=" ..sensorLocTwo
--/?value1=SoundSensor&value2=Kitchen&value3=10:30" 

	  .." HTTP/1.1\r\n" 
      .."Host: maker.ifttt.com\r\n"
      .."Accept: */*\r\n" 
      .."User-Agent: Mozilla/4.0 (compatible; esp8266 Lua; Windows NT 5.1)\r\n" 
      .."\r\n")
	  print("Alert sent to IFTTT.")

end) 

conn:dns("maker.ifttt.com",function(conn,ip) 
    if (ip) then
        print("We can connect to " .. ip)
        conn:connect(80,ip)
    else
		delay()
    end
end)

function delay()
  --wifi.sta.disconnect()
  --wifi.sta.connect()				--wifi.sta.config("","")
  dofile("delay.lua")
end

```

### NodeMCU version
2.2.0

### Hardware
esp8266mod with cp2102

shinji2009, 2018-04-09:
thank you
i tryed to use a separate and it is not working
it is do nothing, last printing is We can connect to 34.227.143.60

```Lua
--ifttt.lua

--wifi.sta.disconnect()
  --wifi.sta.connect()

file.open( "iftttKey.lua", "r" )
iftttKey = file.read()
iftttKey = string.gsub(iftttKey, "%s+", "")
file.close()
--print("IFTTT Maker Key: "..iftttKey)

file.open( "eventName.lua", "r" )
eventName = file.read()
eventName = string.gsub(eventName, "%s+", "")
file.close()
print("IFTTT Event Name: "..eventName)

file.open( "sensorType.lua", "r" )
sensorType = file.read()
sensorType = string.gsub(sensorType, "%s+", "")
file.close()
print("Sensor Type: "..sensorType)

file.open( "sensorLocOne.lua", "r" )
sensorLocOne = file.read()
sensorLocOne = string.gsub(sensorLocOne, "%s+", "")
file.close()
print("Sensor Location: "..sensorLocOne)

file.open( "sensorLocTwo.lua", "r" )
sensorLocTwo = file.read()
sensorLocTwo = string.gsub(sensorLocTwo, "%s+", "")
file.close()
print("Sensor Location: "..sensorLocTwo)

conn = nil
conn2 = nil

conn=net.createConnection(net.TCP, 0) 
conn2=net.createConnection(net.TCP, 0) 
--net.createConnection(type, secure); i.e. Type = TCP,

conn:on("receive", function(conn, payload) 
    --Shutdown!
    gpio.write(4, gpio.LOW)
    tmr.alarm(0, 3000, 1, function()

        delay()

        end)
    end)

conn:on("connection", function(conn, payload) 

print("Sending to IFTTT")
conn:send("POST /trigger/" .. eventName .. "/with/key/" .. iftttKey .. "/?value1=" .. sensorType .. "&value2=" .. sensorLocOne .. "&value3=" ..sensorLocTwo
--/?value1=SoundSensor&value2=Kitchen&value3=10:30" 

	  .." HTTP/1.1\r\n" 
      .."Host: maker.ifttt.com\r\n"
      .."Accept: */*\r\n" 
      .."User-Agent: Mozilla/4.0 (compatible; esp8266 Lua; Windows NT 5.1)\r\n" 
      .."\r\n")
	  print("Alert sent to IFTTT.")

end) 

conn:dns("maker.ifttt.com",function(conn,ip) 
    if (ip) then
        print("We can connect to " .. ip)
        conn2:connect(80,ip)
    else
		delay()
    end
end)

function delay()
  --wifi.sta.disconnect()
  --wifi.sta.connect()				--wifi.sta.config("","")
  dofile("delay.lua")
end

```

TerryE, 2018-04-09:
Please read [our support guidelines](https://nodemcu.readthedocs.io/en/dev/en/support/).  This issues list is _not_ the right place to ask for this type of help.