nodemcu/nodemcu-firmware #2352: UART splitting buffer payload unnecessarily

georeb, 2018-04-12:
When receiving all data from the UART buffer (using 0 as number of bytes to wait for) the first byte comes separately to the remainder of the buffer, making string handling unnecessarily difficult...

### Test code
```Lua
uart.on("data", 0, function(data)
    print("receive from uart: ", data)
end, 0)
```

DATA SENT TO UART: "HelloWorld"

### Expected behaviour

```Lua
receive from uart: HelloWorld
```

### Actual behaviour

```Lua
receive from uart: H
receive from uart: elloWorld
```

### NodeMCU version
Master 2.0

### Hardware
ESP8266 12F

TerryE, 2018-04-12:
[RTFM](https://nodemcu.readthedocs.io/en/dev/en/modules/uart/#uarton).  I/O is non-blocking if the number parameter is zero.  The first on will flush the current buffer into the CB.  How the chars are delivered is up to the hardware, so you need do the buffering in Lua -- unless you use the stop char option, so `'\r'``willl stop on a CR and `HelloWorld\r` will get delivered, but `HelloWorld` will wait for the next `\r`.

closed 2018-04-12 by TerryE

georeb, 2018-04-12:
@TerryE I understand the 'waiting for a character' scenario, as you mentioned, its in the manual. However I don't understand why the hardware would single the first character out?
If I am to use Lua to buffer my data then, what are your thoughts on introducing a `uart.flush()` command to flush the buffer at any given point?

TerryE, 2018-04-13:
> However I don't understand why the hardware would single the first character out?

@georeb, neither do I not why it might stop at the 10th or the 50th or whatever.  If you want to understand then its going to be dependent on your H/W serial I/F, so it's time to get your BitScope out and look at the RX0 timing.  In the case of my Wemos boards, lsusb tells me that they use a "Cygnal Integrated Products, Inc. CP210x UART Bridge" with the CP201x chip handling the virtual USB interface with the host, and this bit-diddles the RX0 line according to the selected baudrate, stop bits, etc. 

I use an SBC for my Home Automation hub and some USB serial connected ESPs to handle all of my relays and sensors.  So I just use the `'\r''`delimiter -- keep it simple stupid.

 If you want to use non-blocking by using  `uart.on("data", 0,)`, then its non-blocking.  Your cb has to handle this.   If you look at the `readline()` code in `luac.c` then you will see that it flushed the uart whenever it is called.   So I don't understand what on earth this extra API call would do.

georeb, 2018-04-13:
> So I don't understand what on earth this extra API call would do.

I have an application where I need to continually listen on the UART and therefore I never recall `uart.on` in which case a `uart.flush()` would be helpful to ensure the UART is clear before attempting to read the buffer.

TerryE, 2018-04-13:
@georeb We shouldn't be having this conversation here. This isn't a bug but more you neither understanding nor accepting how the SDK and the firmware architecture works.  It would be helpful if I were a multi-millionaire, but I am not.  I will say it again.  The `on('sent',0)` always flushes the UART fifo.  If you want to ignore this programmatically then you can.  The SDK isn't a pre-emptive OS.  It is (in principle) non-blocking and event driven.  The UART is handled by an ISR asynchronously.  Your Lua get scheduled as a task.  If you need or want flow control handshake then you need to implement this at an application level.  If the other end of there serial interface wants to do flood xfer, then no matter what you do at an applications level, the UART fifo will collect input in parallel to you sending a response.  **_So I don't understand what on earth this extra API call would do._**

pjsg, 2018-04-13:
The reason that the first character is a single character is due to the way that UARTs work. When the first character is received, an interrupt is generated and the interrupt handler essentially posts a task to get the data and trigger the callback. If the system is idle, then this happens before the second character is received. The interrupt handler keeps doing this and by the time the system goes idle again, and the task is processed, lots more characters have arrived.

I predict that if you modified your test code and kept the system busy, then you would sometimes see more than one character arriving in the first callback.

georeb, 2018-04-13:
Misunderstanding is exactly the reason I am asking. Please have a little more patience.
However, you're right, I should take this query elsewhere, esp8266.com forum perhaps.

georeb, 2018-04-13:
> I predict that if you modified your test code and kept the system busy, then you would sometimes see more than one character arriving in the first callback.

Great, thanks @pjsg I shall try that out!