nodemcu/nodemcu-firmware #2354: ESP32: GATT support?

chengyuhui, 2018-04-16:
After searching around I have noticed that none of the lua SDKs around have a GATT API. Is there any plan to support this?

devsaurus, 2018-04-16:
There are no plans currently. PRs are welcome.

chengyuhui, 2018-04-17:
@devsaurus Thanks. Will try to work on that after the upcoming exam.

chengyuhui, 2018-04-19:
A question: Is `lua_getstate()` reliable for use? The bluetooth stack does not have a way to pass user data to callbacks, `bthci` uses it, but the function is not documented anywhere.

devsaurus, 2018-04-19:
Yes, it's even preferred over caching a local state.
The code base was globally ported to `lua_getstate()` with #1423, the related issues provide some technical details.

chengyuhui, 2018-04-21:
Just set up the building environment,

Building works fine, but I'm getting `function call is not allowed in a constant expression` in module files about those `LROVAL` macros when developing in VSCode?

chengyuhui, 2018-04-22:
When will those `xxx_init` function be called? Is it apporiate to initialize bluetooth stack there?

devsaurus, 2018-04-22:
They are called once during setup of the Lua VM before any script or function is executed. Very similar to callbacks which are executed after system reset before the main loop starts.
They're an appropriate place to perform one-off initialization of drivers or stacks for example.

chengyuhui, 2018-04-22:
Is there any way to declare that my module conflicts with `bthci`, which initializes the BT driver differently?

devsaurus, 2018-04-22:
You could model the mutual exclusion in menuconfig like it's done for the panic handler with `choice`:
https://github.com/espressif/esp-idf/blob/9b7454de0c171a7f94dd8f42a5453ef6c75370e0/components/esp32/Kconfig#L476
Instead of enabling `bthci`, there could be a choice to select either `bthci` or the GATT module.

chengyuhui, 2018-05-02:
I've noticed that if the bluetooth stack is enabled, the firmware is not bootable (too large) unless a custom partition table is used, what should I do? Simply document this?

devsaurus, 2018-05-04:
> I've noticed that if the bluetooth stack is enabled, the firmware is not bootable (too large)

Do you refer to the size of `build/NodeMCU.bin`? I get something around 760kByte max with bluetooth module at the moment. Far from 1&nbsp;MByte.

Do you have any indication what exactly drives the size increase? Is your work in progress available on github?

> Simply document this?

Once we know what the max firmware requirement is, we could provide tailored custom tables for 2&nbsp;MByte and 4&nbsp;MByte modules. E.g. the current template at `components/platform/partitions-2MB.csv` just artificially restricts the firmware to 1&nbsp;MByte and leaves 960&nbsp;kByte to the SPIFFS partition.

chengyuhui, 2018-05-05:
@devsaurus It is available in my fork (there is a gatts branch)

I simply tested with bluetooth enabled or disabled, the only difference between default config is than I also have `ws2812` module enabled.

devsaurus, 2018-05-11:
> It is available in my fork (there is a gatts branch)

Can't compile that atm.

chengyuhui, 2018-05-11:
@devsaurus A latest commit was missing, I currently cannot access my computer, will upload tomorrow.

chengyuhui, 2018-05-12:
The code has been updated.

devsaurus, 2019-05-31:
Feel free to re-open or submit PR.

closed 2019-05-31 by devsaurus