nodemcu/nodemcu-firmware #2361: wifi.suspend() crashes

eyaleb, 2018-04-28:
### Expected behavior

wifi.suspend(a,b,c,d) suspend WiFi for a nominated period.
wifi.suspend() should return the status.

The context is this: my app automatically gets a wifi connection (`STA_GOT_IP`) very quickly, normally within 500ms (as tmr.now() reports). However, at times it takes 2.5 seconds, probably due to clashing with the large number of active devices around (about a dozen connecting every minute).

I added a timeout alarm if the connection takes longer that 1000ms when I want to shut down wifi and resume it immediately (looks like 50ms is the shortest period allowed), thus connecting quicker (I do not know that this will be any faster).

### Actual behavior

Any call to wifi.suspend() produces this:

suspend WiFi

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26532, room 16 
tail 4
chksum 0x8e
load 0x3ffe8000, len 2180, room 4 
tail 0
chksum 0x14
load 0x3ffe8884, len 136, room 8 
tail 0
chksum 0x62
csum 0x62

NodeMCU 2.2.0.0 build dev-20180414-1309-1m-all powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)

### Test code

```Lua
local function dummy() end
print ("suspend WiFi")
wifi.suspend({50000, dummy, dummy, true})
```
or even just this:

```lua
print ("reset WiFi, state=%d", wifi.suspend())
```

### NodeMCU version

Using dev branch off git, as of 14/Apr.

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

Any, but in this case it is an esp-12.

eyaleb, 2018-04-29:
I created a [small program](https://github.com/makehackvoid/ESP8266Lib/blob/master/examples/eyal/iw.lua) to demonstrate the situation.

Sample output. Initially all is well for a while
```
NodeMCU 2.2.0.0 build dev-20180414-1309-1m-all powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
0.172506 sta.status=1 sta.getip=192.168.2.52/255.255.255.0
0.173926 registering eventmons
0.177094 set main  wifi timeout 1000ms
> 0.187219 AUTHMODE_CHANGE
0.249882 STA_CONNECTED
0.262289 STA_GOT_IP
0.271401 have connection sta.status=5 sta.getip=192.168.2.52/255.255.255.0
0.279157 deep sleep 2s
```

The above repeats until a slower connection leads to this
```
NodeMCU 2.2.0.0 build dev-20180414-1309-1m-all powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
0.181576 sta.status=1 sta.getip=192.168.2.52/255.255.255.0
0.183004 registering eventmons
0.186172 set main  wifi timeout 1000ms
> 0.196437 AUTHMODE_CHANGE
1.193051 connection timeout, status=1
1.199903 retry 1^M<FF>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26532, room 16 
tail 4
chksum 0x8e
load 0x3ffe8000, len 2180, room 4 
tail 0
chksum 0x14
load 0x3ffe8884, len 136, room 8 
tail 0
chksum 0x62
csum 0x62
```
The crash (which then repeats indefinitely) is on this line
```
Log ("suspend status=%d", wifi.suspend())
```
HTH

dnc40085, 2018-04-29:
`wifi.suspend({50000, dummy, dummy, true})`
should be:
`wifi.suspend({duration=50000, suspend_cb=function() print("wifi suspended") end, resume_cb=function() print("wifi suspended") end, preserve_mode=true})`
or
```lua
cfg={}
cfg.duration=50000
cfg.suspend_cb=function() print("wifi suspended") end
cfg.resume_cb=function() print("wifi restored") end
cfg.preserve_mode=true
wifi.suspend(cfg)
```

eyaleb, 2018-04-29:
Yes, now fixed.

However note that the crash is earlier, for the call without arguments that is supposed to return the status.

eyaleb, 2018-04-29:
```
NodeMCU 2.2.0.0 build dev-20180414-1309-1m-all powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
0.178851 sta.status=1 sta.getip=192.168.2.52/255.255.255.0
0.180281 registering eventmons
0.183427 set main  wifi timeout 1000ms
> 0.229614 DISCONNECTED reason 201
0.239550 retry 1^M<FF>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 26532, room 16 
tail 4
chksum 0x8e
load 0x3ffe8000, len 2180, room 4 
tail 0
chksum 0x14
load 0x3ffe8884, len 136, room 8 
tail 0
chksum 0x62
csum 0x62

NodeMCU 2.2.0.0 build dev-20180414-1309-1m-all powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)
        reset_reason:2 EXCCAUSE:3 EPC1:40241EA4 EPC2:0 EPC3:0 EXCVADDR:40265D4D DEPC:0
```

eyaleb, 2018-05-06:
Thanks @dnc40085, with the config option it now works.

Sadly, cheating the wifi does not work: if I deep sleep (or wifi suspend) before a connection is established then on wakeup the connection takes much longer (over 2.5s instead of under 300ms).
Looks like either the esp or the a/p do some sort of a cold-start connection.

PR #2365 (2018-05-07): Bugfix: wifi.suspend() disabled message causing crash
Fixes #2361.

- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.

Issue #2361 brought to my attention a mistake I made when writing the code that tells the developer that `wifi.suspend()` is disabled when `PMSLEEP_ENABLE` is commented out in `app/include/user_config.h`.
closed 2018-05-06 by eyaleb