nodemcu/nodemcu-firmware #2362: [ask] is it safe for wifi?

johndoe71rus, 2018-04-29:
I'm trying to make a template for my devices. I do not remember where I found this code on the network. Esp8266 creates an AP, on the page you can select the available network wifi and connect. If there is a power failure, there are situations when the esp created its AP and at the same time it is available in the home network. 2 connections are created. it is safe? A hacker can connect to an esp ap and then explore the esp to find out the password from the home network?

commit: 2e67ff5a639a13260fd4fb3c3b627ccdc2845616
SSL: false
modules: adc,bme280,file,gpio,i2c,net,node,ow,spi,tmr,uart,wifi
build built on: 2017-09-28 04:15
powered by Lua 5.1.4 on SDK 2.1.0(116b762)

https://paste.ubuntu.com/p/5pVP95q3Jw/ initsens.lua
https://paste.ubuntu.com/p/nKnkZRCgVk/ configServer.lua
https://paste.ubuntu.com/p/tpvCJRGcDW/ configServerInit.lua

TerryE, 2018-04-29:
Please read our [support policy](https://nodemcu.readthedocs.io/en/master/en/support/) for this type of Q.

closed 2018-04-29 by TerryE

johndoe71rus, 2018-04-30:
OK. So, in your opinion, working with two wi-fi networks at the same time is not a potential vulnerability

TerryE, 2018-04-30:
That's a bit of bizarre induction.  No, we have a scope for issues posted here and we have templates; you have ignored both, so the issue was closed.

As to the vulnerability, the functionality is available through the SDK and can be equally expressed in a standard C over SDK implementation.  We don't see our job to try to add layers of pseudo trust, so your point doesn't relate to an bug in the firmware.

IMO anyone who runs IoT infrastructure on anything other than an isolated control network opens themselves up to direct attack, yet this seems to be the trend with many IoT devices directly access cloud services.

If  you want to raise an issue here, then follow our posting guidelines.