nodemcu/nodemcu-firmware #2363: Lua strings truncated by nullchar, despite being supposedly 8-bit clean

georeb, 2018-04-30:
> The manual wants to make it clear that a '\0' is not a terminator in Lua. It also means that you can load arbitrary bytes into a string (image, audio, video, native code, etc.) and you do not risk having it truncated at the first '\0' by some library function (which could happen in C if you use string-related functions).

(src: [StackOverflow](https://stackoverflow.com/questions/8475656/whats-an-embedded-zero))

The Lua 5.1 manual states...

> String represents arrays of characters. Lua is 8-bit clean: strings can contain any 8-bit character, including embedded zeros ('\0')

(src: [Lua.org](https://www.lua.org/manual/5.1/manual.html#2.1))

However, this does not appear to be the case.

### Test code

```Lua
print(string.char(97, 98, 99, 0, 100, 101, 102))

-- expected output: abc def
-- actual output: abc
```
### NodeMCU version
Master

### Hardware
ESP8266 12F

georeb, 2018-04-30:
@TerryE - as requested.
First noticed while using the `http.get()` module.

TerryE, 2018-04-30:
@georeb, the offending code is in `lbaselib.c:luaB_print()`.  I've checked standard Lua .5.1.5 and this extracts the strings and issues a `fputs(s, stdout)`.  NodeMCU Lua uses pretty much the same code with `c_fputs(s, c_stdout)`, though interestingly Lua 5.3.4 replaces this with `lua_writestring(s, l)` where `l` is the length of the Lua string.

In other words all Lua 5.1 versions have the same bug: the print treats Lua strings as Cstrings. Bugger.  

All I can suggest is to use `uart.write(0, ... )` though note:
-  parameters can be strings or numeric. 
-  strings are output full size, that is including any embedded 0s
-  numbers are treaded as string.char(n)
-  print separates fields by a `'\t'` char, but write doesn't so you need to add your own field separators if needed.
-  likewise no CRLF terminator is added.

Since `uart.write()` concatenates parameters for output then `write(a,b,c)` is always more efficient than the `a..b..c` variant, especially if you've cached `local write = uart.write` before hand.

So I hope this answers your issue. I will close this because our upgrading to Lua 5.3 will fix this anyway and we won't break backwards compatibility within a Lua version.  Thanks for pointing this out.

closed 2018-04-30 by TerryE

georeb, 2018-05-01:
Thanks for investigating this @TerryE , however it was not the `print` function that I was trying to use (although that is clearly affected). I was trying to write the data received from an `http` get request to a file. So even when not printing, the Lua strings are still truncating, as only part of the string (up to the first null) was written to the file. Hope this makes sense.

pjsg, 2018-05-01:
I have certainly sent nulls to a file -- what exactly are you doing?

georeb, 2018-05-01:
Please see below @pjsg ...

```Lua
cnksize=4096
url="http://www.myserver.com/"

function getfile(fnm)
	file.remove(fnm)
	http.request(url..fnm, "HEAD", nil, nil, 
	function(code, data, headers)
		if (code < 0) then
			print("Failed to get file: "..fnm)
		else
			for k, v in pairs(headers) do
				if string.lower(k) == "content-length" then
					print("File: "..fnm)
					print("Length of file: "..v.." bytes")
					print("Getting file...")
					getdata(fnm,1,math.ceil(v/cnksize))
				end
			end
		end
	end)
	
end

function getdata(fnm,cnk,noc)
	
	frombyte = (cnk*cnksize)-cnksize
	tobyte = frombyte+(cnksize-1)
	
	print("Getting chunk "..cnk.." of "..noc.."("..frombyte.."-"..tobyte..")".." - "..math.ceil((100/noc)*cnk).."%")
	
	http.get(url..fnm, "Range: bytes="..frombyte.."-"..tobyte.."\r\n", function(code, data)
		
		print("Payload length: "..#data)
		--print("Payload: "..data)
		
		if (code < 0) or #data~=cnksize then
			print("Failed to get file data at chunk: "..cnk)
		else
			file.open(fnm,"a")
			file.write(data)
			file.close()
			if cnk<noc then
				cnk = cnk+1
				getdata(fnm,cnk,noc)
			else
				print("Got all chunks and file saved!")
			end
		end
	end)
	frombyte = nil
	tobyte = nil
end

getfile("myfile.txt")
```

TerryE, 2018-05-03:
@georeb, George, you can just write NULLs to a file no problem.  You are asking the wrong Q.  Somewhere in your code and APIs use Lua strings are being interpreted a Cstrings.  You real Q is: exactly where is this happening?  Set up a simple test case and do a binary chop to find out.

HHHartmann, 2018-05-06:
Just read your sample and I would suggest to use GET instead of HEAD in the request.
At w3.org it states that the HEAD methode "MUST NOT return a message-body" so I would not expect any data here but maybe your server sends some data instead, but not all.

TerryE, 2018-05-06:
@HHHartmann Gregor, if you read the example then you will see that George uses the HEAD request correctly, and gets the content as a set of data chunks.

HHHartmann, 2018-05-07:
@TerryE  of course you are right. I just saw the HEAD at the beginning and the loop at the end, missing on the http.get in the middle ...

georeb, 2019-07-29:
@TerryE Please can I suggest that this issue is reopened as the same issue exists in today's latest master branch!

```
print(string.char(97, 98, 99, 0, 100, 101, 102))

-- expected output: abc def
-- actual output: abc
```

TerryE, 2019-07-29:
@georeb, try exactly the same thing on any lua 5.1 distro.  I just ran this on my Linux laptop:
```
$ lua51
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio
> print "ab\0cd"
ab
>
``
It's how Lia 5.1 handles printing strings! Lua5.3 does it properly.`

georeb, 2019-07-29:
@TerryE I thought we'd already upgraded to 5.3 - not sure why, sorry its been a long day!
Looking forward to the update 👍

TerryE, 2019-07-29:
We're in the middle of the work which is why you might be seeing lots of issues and PRs related to this :smile: