nodemcu/nodemcu-firmware #2364: MQTT  memory leak, reconnection

bondrogeen, 2018-05-03:
A memory leak occurs during reconnection. Until the memory is over.

### Expected behavior

### Actual behavior
LOG
```Lua
> Connected to MQTT...
failed reason: nil
36592
failed reason: -5
36480
failed reason: -5
36368
failed reason: -5
36256
failed reason: -5
36144
failed reason: -5
36032
failed reason: -5
35920
failed reason: -5
35808
failed reason: -5
35696
failed reason: -5
35584
Connected to MQTT...
=node.heap()
35080
> =node.heap()
35080
> failed reason: nil
35184
failed reason: -5
35472
failed reason: -5
35360
failed reason: -5
35248
Connected to MQTT...

```
### Test code
```Lua
m = nil
mqttReconnectTimer = tmr.create()
function handle_mqtt_error(client, reason)
 client:close() 
 mqttReconnectTimer:start()
 print("failed reason: "..tostring(reason))
 print(node.heap())
end
function handle_mqtt_connect(client)
 print("Connected to MQTT...")
 client:on("offline", handle_mqtt_error)
 client:on("message", function(c,t,d) end)
end

function mqtt_connect(client)
 client:close()
 client:connect("192.168.1.61", 1883,0, handle_mqtt_connect, handle_mqtt_error)
 collectgarbage()   
end
function start()
 m = mqtt.Client("ESP", 120, "admin", "12345")
 mqttReconnectTimer:register(1000, tmr.ALARM_SEMI, function() 
  mqtt_connect(m) 
 end)
 mqttReconnectTimer:start()
end

start()
```
### NodeMCU version
branch: dev and master

NodeMCU custom build by frightanic.com
	branch: dev
	commit: 3ec252df6a08eb087fc4d01bfd6719db6475fc43
	SSL: false
	modules: adc,bit,crypto,ds18b20,file,gpio,http,i2c,mqtt,net,node,ow,pwm,rotary,rtctime,sjson,spi,tmr,uart,wifi
 build created on 2018-05-03 15:05
 powered by Lua 5.1.4 on SDK 2.2.1(cfd48f3)

### Hardware
ESP-12, wemos d1 mini

marcelstoer, 2018-05-07:
Looking at the log one can conclude that successfully establishing a connection after several failed attempts does free some memory. Once "Connected to MQTT..." is printed the heap _increases_ for a few cycles.

xfoldvar, 2018-05-07:
A successful connection consumes about 500 bytes of heap and this amount of memory is partially freed when connection is lost, but each reconnect attempt consumes 112 bytes, which are never freed - even if connection is reestablished.

devsaurus, 2018-05-08:
I can confirm the leak. My testcase is along the example above and tries to connect to an unresponsive server IP.

So far, the issue seems to be related to closing the espconn connection when the connection timeout elapsed. I try to understand how espconn requires the application to gracefully shutdown without memory leak when there never has been a connection at all (current mqtt module code just uses `espconn_disconnect()`).