nodemcu/nodemcu-firmware #2366: can't scroll docs

konsumer, 2018-05-07:
The [documentation site](http://nodemcu.readthedocs.io) makes it impossible to read anything that is below the scroll-line (for example: [here](http://nodemcu.readthedocs.io/en/master/en/build/)) on chrome `Version 66.0.3359.139 (Official Build) (64-bit)`, so I can't scroll through a doc-page. I tested on incognito, to ensure the issue isn't related to a plugin or something else. The readthedocs site appears to have the same issue with other projects.

This is not an issue with this project, but it prevents me from reading the docs without extracting the HTML.

### Expected behavior

The page should scroll through all the content.

### Actual behavior

It doesn't scroll, so any content longer than a page gets cut off. The sidebar seems unaffected.

pjsg, 2018-05-07:
Wow! Chrome is definitely broken. I checked in Safari and it worked fine.

pjsg, 2018-05-07:
Do you have the experimental features enabled (go to chrome://flags ) -- there is a bug report that indicates that this is the problem.

konsumer, 2018-05-07:
Aha! great troubleshooting. That was totally it. Works great now. Thanks!

closed 2018-05-07 by konsumer