nodemcu/nodemcu-firmware #2370: Packet loss

Lir10, 2018-05-10:
Using the `net `module i'm sending packets every 1 second from the server to the ESP 

From a total of 1000 packets i can see that i lost 14 packets.

I can see with WireShark that all the packets have been sent succesfuly but i can't see those packets in the net Receive callback

Any ideas?

Thanks.

marcelstoer, 2018-05-10:
Are you absolutely certain they weren't lost along the way _before_ they reach the MCU? Are you saying the MCU might consume those packets w/o triggering the callback?

Lir10, 2018-05-10:
Hi @marcelstoer , that's exactly what i'm seeing - it's very likely that that MCU does consume those packets w/o triggering the callback

Lir10, 2018-05-10:
@marcelstoer Currently i have only confirmed this with Wireshark on the server side. is there any tool that can help us to prove it on the MCU ?

TerryE, 2018-05-14:
If your connection is TCP, then this is serious, however if UDP then

> UDP is a minimal message-oriented transport layer protocol that is documented in RFC 768. UDP provides no guarantees to the upper layer protocol for message delivery and the UDP layer retains no state of UDP messages once sent. For this reason, UDP sometimes is referred to as _Unreliable_ Datagram Protocol

Lir10, 2018-05-14:
@TerryE it's TCP

TerryE, 2018-05-14:
Ah, that is an issue since LwIP should handle the assured delivery.  So for the avoidance of doubt, what you are saying is that if you do a 1,000 packet transfer over TCP then you will typically drop 3 packets at the receiving end.  Can you give us a minimal gist test case of both the client and server ?

closed 2019-08-04 by stale[bot]