nodemcu/nodemcu-firmware #2371: PIN_FUNC_SELECT by lua?

anod221, 2018-05-14:
My problem is that I need a pin for clock out. I found that GPIO0 would be work. [pin func](https://www.esp8266.com/wiki/doku.php?id=esp8266_gpio_pin_allocations)
So I think I need to call PIN_FUNC_SELECT by lua. Is there any way to do that?

devsaurus, 2018-05-17:
I'm not too comfortable with introducing a dedicated mode for just a single pin (and ensure that it fails when applied to all other pins). Though I don't have better proposal to abstract this.
Something like `gpio.FUNCTION4` would be more generic, but is really low-level and requires the user to select the matching function for the intended mode.

anod221, 2018-06-04:
OK, I think I should do it myself. Thanks.

closed 2018-06-23 by marcelstoer