nodemcu/nodemcu-firmware #2372: Pin declared in IDE does not match output blink number (Noob!)

fritserasmus, 2018-05-14:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior : I was expecting that if the LED is connected to GPIO2/D4 the LED would blink

**A brief summary of my IDE**
int RedLEDPin=5;

void setup ()  {
    pinMode(RedLEDPin,OUTPUT);

void loop ()  {
   digitalWrite(RedLEDPin,HIGH);
   delay(waitTimeOnF);
   digitalWrite(RedLEDPin,LOW);

### Actual behavior
After trail and error PIN 2 (D1 / GPI05) blinks

Board: NodeMCU 0.9 (ESP-12 Module)
Baud: 115200 

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
Arduino 1.8.5

```
int RedLEDPin=5;
int waitTimeOnF=200;
int waitTimeOffF=200;

void setup ()  {

pinMode(RedLEDPin,OUTPUT);

}

void loop ()  {

digitalWrite(RedLEDPin,HIGH);
delay(waitTimeOnF);
digitalWrite(RedLEDPin,LOW);
delay(waitTimeOffF);

}

```
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

LoLin ESP8266 NodeMCU 
new ModeMcu V3 

8<------------------------ END BUG REPORT -------------------------------------

FrankX0, 2018-05-14:
This repository is for Lua based firmware.
Your request is for the Arduino firmware..
Please close this issue.

fritserasmus, 2018-05-14:
Apologies Franc,
I thought I need the same firmware for Arduino and Lau

closed 2018-05-14 by fritserasmus