nodemcu/nodemcu-firmware #2376: file.read(): On end of file, rather than nil, nothing is returned

dnc40085, 2018-05-19:
### Expected behavior
The functions [`file.read()`](http://nodemcu.readthedocs.io/en/dev/en/modules/file/#fileread-fileobjread) and [`file.readline()`](http://nodemcu.readthedocs.io/en/dev/en/modules/file/#filereadline-fileobjreadline) should [return `nil`](http://nodemcu.readthedocs.io/en/dev/en/modules/file/#returns_14) on subsequent calls after end of file has been reached.
### Actual behavior
After end of file is reached, nothing is returned on subsequent calls.
### Test code
```Lua
do
  file.open("init.lua", "r")
  while true do 
    retval=type(file.read())
    print(retval)
    if retval=="nil" then print("this never happens") break end
  end
  file.close()
  print("End (This also never happens)")
end
```
This code never reaches the end, it just errors out with:
`stdin:5: bad argument #1 to 'type' (value expected)`
### NodeMCU version
 dev( 914a4af)
### Hardware
ESP12-F

### Fix 
in `/app/modules/file.c` in the function `file_g_read()` on line [435](https://github.com/nodemcu/nodemcu-firmware/blob/914a4afc96136311db8ab923b8a6942f429d266f/app/modules/file.c#L435) replace:
```c
return 0;
```
with:
```c
lua_pushnil(L);
return 1;
```
Unless the documented return value is in error, I can submit a pull request with this fix.

TerryE, 2018-05-19:
The string "nil" is not the same as the Lua type `nil`. Try `if not retval then print("this never happens") break end`

djphoenix, 2018-05-19:
@TerryE look for call to “type” function between file.read and comparation.

dnc40085, 2018-05-19:
>The string "nil" is not the same as the Lua type nil.

@TerryE I think you didn't look close enough at the test code...
According to the [documentation for this function](http://nodemcu.readthedocs.io/en/dev/en/modules/file/#fileread-fileobjread), the return values for `file.read()` and `file.readline()` should be **File content as a string, or nil when EOF**.
Therefore, on end of file(EOF) the return for `type(file.read())` should be a string `"nil"`

Please @TerryE , give the test code a try, I'm sure you'll find that it doesn't even get to the `if` test.

TerryE, 2018-05-19:
@dnc40085, my apologies, you are correct that `file.read()` returns an empty result list, rather than a single nil value, but in fact this is a _very_ common Lua convention which exploits Lua assignment rules.  If the file is at EOF and `file.read()` is evaluated in a scalar context then this returns `nil`.
 grep -PR '\breturn\s+0\b' .

dnc40085, 2018-05-19:
>From a performance and code size perspective doing a type(file.read()) makes absolutely no sense 

I agree, I only used `type(file.read())` to get an example that would reproduce the issue.

I was actually debugging something that read a file 1024 chars at a time.
I was stepping through the file with `print(file.read())` and when I got to EOF it was printing blank lines rather than nil, which led me to believe that `file.read()` was returning `""` instead of `nil` which didn't match up with the documentation, so I tried `type(file.read())` and got `(value expected)`.

Before opening this issue I figured out that it wasn't returning a blank string and that the problem was elsewhere in my program, but I initially was confused by the lack of the `nil` return and I figured that I might try prevent somebody else down the line from getting confused as well. 

I don't see it as a bad thing to exploit the Lua assignment rules, I imagine doing so reduces memory usage and cuts down on firmware size, Having more memory is never a bad thing, especially when developing. On a side note I would like to add that I have found your LFS PR very useful and has been working without issue since I started using it a little over a week ago, Thank you for all your work on the LFS update.

PR #2383 (2018-05-27): Fix for lack of nil return in file.read() when EOF is reached
Fixes #2376.
- [X] This PR is for the `dev` branch rather than for `master`.
- [X] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [X] I have thoroughly tested my contribution.

This PR adds the missing `nil` return when a file is stepped through with `print(file.read())` and the end of file is reached.
closed 2018-05-27 by TerryE