nodemcu/nodemcu-firmware #2377: Memory use with file.readline()

s-wacker, 2018-05-20:
### Expected behavior
Read line, no permanent memory consumption.

### Actual behavior
Readline works, but causes permanent memory consumption.

Console output directly after startup:
>=collectgarbage("count")
4

Function is run via console:
> increment_job_counter()

Memory usage increases:
> =collectgarbage("count")
5

> increment_job_counter()
> =collectgarbage("count")
6

> increment_job_counter()
> =collectgarbage("count")
7

> collectgarbage()
> =collectgarbage("count")
7

This example can be continued until memory overflow error occurrs.

### Test code
This is my minimalistic init.lua
```Lua
function increment_job_counter()
	if file.exists("job.counter") == true then
		local handle = file.open("job.counter", "r")
		local val = tonumber(file.readline()) + 1
		handle = file.open("job.counter", "w+")
		handle:writeline(tostring(val))
		handle:close()
		handle = nil
		val = nil
	else
		local handle = file.open("job.counter", "w+")
		handle:writeline(tostring(1))
		handle:close()
		handle = nil
	end
end
```
### NodeMCU version
branch: master (latest)
commit: 67027c0d05f7e8d1b97104e05a3715f6ebc8d07f

### Hardware
Wemos D1 mini pro

TerryE, 2018-05-22:
OK, I had a binary chop on this and the simplest code which demonstrates the leak is
```Lua
do 
     file.open("job.counter", "r")
     for i = 1,10,1 do
          file.readline()
--          file.seek "set"
          print("used memory: ",collectgarbage("count")*1024)
     end
     file.close()
end
```
It doesn't matter whether you reseek to pos 0 or not; it's the readline that leaks.  Replacing this with `file.read(1)` works fine.  I'll take a look at the C code.

BTW your `tonumber(file.readline())` should be `tonumber(handle:readline())`

TerryE, 2018-05-22:
Another update.  I added `node.heap()` to the "memory used" diagnostic and ran this for 100 iterations.  The last few are:
```
used memory:    105421  Heap:   42552
used memory:    106444  Heap:   42552
used memory:    107467  Heap:   42552
````
We only have ~48Kb heap on the ESPs and `node.heap()` is reporting a solid 42552; it's the LGC reported value that is crap.

Now the file read routine does a:
-  `luaM_malloc(L, FILE_READ_CHUNK)` which is a macro for `luaM_realloc_(L, NULL, 0, 1024)` followed by
-  `luaM_free(L, heap_mem)` which is a macro for `luaM_realloc_(L, NULL, sizeof(char), 0)`

The LGC counting code uses the 3rd and 4th parameters as the start and end lengths and these would need to be (0,1024) and (1024,0) to net out to zero, but the second pair are (1,0) so the LGC count loses 1023 on each cycle.  This is an audit error only and not a memory error, per se. 

Bugger

TerryE, 2018-05-22:
Just to validate this, I replaced the `luaM_free(L, heap_mem)` calls by `luaM_freearray(L, heap_mem, n0, char);` where n0 is the size of the buffer allocated and bingo, the error goes away.

I don't want to create a PR for this one error, though others are welcome to.  What we really need to do here is to go through all uses of `luaM_free()`  in our code and put in a bulk fix of the lot.

The other reason is that the current implementation of `file_g_read()` is really an exemplar of how to write terribly inefficient code and it desperately needs rewriting if I touch it properly -- which I just don't have time to do at the moment.

TerryE, 2018-05-22:
Rather than close this, I've renamed it to reflect the actual bug, so we can leave this open and track it.

@s-wacker, thanks for finding and reporting this.

cwrseck, 2018-05-23:
Something's odd here - running your code on an integer
version of NodeMCU 2.1.0-master_20170824 I get:

> dofile("newrd.lua")
used memory: 	5658
used memory: 	5660
used memory: 	5660
used memory: 	5660
used memory: 	5660
used memory: 	5660
used memory: 	5660
used memory: 	5660
used memory: 	5660
used memory: 	5660
> 

Running the following code:

do 
    file.open("job.counter", "r")
    for i = 1, 10, 1
    do
        file.readline()
        -- print("used memory: ", collectgarbage("count") * 1024)
        local h = node.heap()
        print("Heap space (b):    " .. h)
        local k, b = collectgarbage("count")
        k = k or 0
        print("Memory usage (kb): " .. k)
    end
    file.close()
end

I get:

> dofile("newrd.lua")
Heap space (b):    42024
Memory usage (kb): 6.673828125
Heap space (b):    41960
Memory usage (kb): 6.673828125
Heap space (b):    41960
Memory usage (kb): 6.673828125
Heap space (b):    41960
Memory usage (kb): 6.673828125
Heap space (b):    41960
Memory usage (kb): 6.673828125
Heap space (b):    41960
Memory usage (kb): 6.673828125
Heap space (b):    41960
Memory usage (kb): 6.673828125
Heap space (b):    41960
Memory usage (kb): 6.673828125
Heap space (b):    41960
Memory usage (kb): 6.673828125
Heap space (b):    41960
Memory usage (kb): 6.673828125
> 

Why does the FP / Integer change affect the output?

Will

TerryE, 2018-05-23:
@csreck, I checked the 1.5.4-1-final branch and the same combination is there, except that this commit 442e3e2e on 19 Dec 2017 replaced the lookahead size for readline() from `LUAL_BUFFERSIZE` to `FILE_READ_CHUNK` because the docs state that the lookahead size is 1024 bytes, but  `LUAL_BUFFERSIZE` had been dropped from 1024 to 2056 in f9533ed on 16 Oct 2016 because under some circumstances this could save a lot of C stack space (and remove a high percentage of crash due C stack overrun).

However if the lookahead <= `LUAL_BUFFERSIZE` then `alloca()` was used so the 19 Dec 2017 change cause the `file.c` code to take a different execution path, thus exposing this error.

TerryE, 2018-10-18:
> The wider question is should we really be using the `luaM_*` functions at all?

Yes for three reasons:
-  The Lua GC has a lazy element and so a LuaM alloc operation will rerun the Lua GC and retry whereas the `os_` functions just fail and return 0.
-  The LuaM operations throw an error on failure
-  The `collectgarbage()` rely on the luaM functions being used and used correctly.   

The main point is that you need to pair the alloc and free types correctly  so that you either use the object from for both or the vector form for both for the usage stats to be calculated correctly.

tomsci, 2018-10-18:
Sorry I should have been clearer as to what I meant - I wasn't objecting to leveraging Lua's memory management functions, I totally get what benefits they provide. My issue is that `luaM_free()` looks like it can correctly free any pointer returned by `luaM_malloc()` regardless type (because that's how `malloc()` and `free()` behave) - but it can't, and pretty much every use of it I can find in nodemcu assumes that it can. For untyped `char*` pointers one must call `luaM_freemem()` also passing in the size of the allocation. My musings therefore were towards should there be some sort of wrapper around `luaM_malloc()`/`luaM_free[mem]()` which make it less likely people will make the mistake of thinking those functions work like malloc and free.

jpeletier, 2020-01-29:
Don't close this, still work in progress

HHHartmann, 2021-01-24:
No progress but still TODO

tomsci, 2021-01-24:
In case it is useful, this is a commit that fixes a couple of occurences of this issue in node.c: https://github.com/tomsci/nodemcu-firmware/commit/7cb19143f3e867e3ecdce424a5898e59b028ed2f

closed 2022-04-30 by stale[bot]