nodemcu/nodemcu-firmware #2381: Building the Espressif SDK 2.2 cross-compiler

cwrseck, 2018-05-26:
It might be worth summarising the ESP8266 SDK build process somewhere in
the documentation - pfalcon/esp-open-sdk is well behind the times, and
synchronising crosstool-NG and the ESP SDK isn't easy.  In particular,
it's not clear where to find the current libc.a, libhal.a, and include.tgz
iles, and which patches to add, for the 2.2 SDK.

(If I ever work it out, I'll update this.)

Will

TerryE, 2018-05-26:
@cwrseck Will,  the current make unpacks and uses the minified toolchain [tools/esp-open-sdk.tar.xz](../blob/master/tools/esp-open-sdk.tar.xz) which is a gcc  version 4.8.5 build _if_ you haven't already got a toolchain installed and pathed in.  The make also download and installs the current 22 SDK.  The SDK has a higher search precedence than the toolchain, so

-  whilst there are 2 copies of `libc`, if you search the `app/mapfile` then you will see  that the current SDK version is used
- the toolchain version of `libhal.c` is used.

cwrseck, 2018-05-27:
Thanks - your note makes things much clearer, and would be well worth adding to the FAQ.

Will

closed 2018-07-07 by TerryE