nodemcu/nodemcu-firmware #2403: Full math library on nodemcu-build.com

alexandruantochi, 2018-06-21:
### Missing feature
math library only has abs, ceil, floor, max, min, pow, random, randomseed, sqrt, pi, huge, Infinity

### Tell us why you would like to see this feature added.

I need to calculate decibels using a log10 function. I see that there is a full math library at 

https://github.com/nodemcu/nodemcu-firmware/blob/master/app/lua/lmathlib.c
### Workarounds
no

TerryE, 2018-06-21:
This was done back in the day when most ESP module had 512Kb Flash.  It would be trivial to add a  `user-config.h` option to enable this.  We have to be a little careful to check these extra libraries load into flash and not ram.

marcelstoer, 2018-06-23:
I fully support this but it should be disabled by default and would also make the flag available in my cloud builder. @alexandruantochi do you know about @1010? Have you considered putting a PR up yourself?

alexandruantochi, 2018-06-23:
@marcelstoer  Sorry I don't follow regarding @1010. I could put a PR in the future, but I'm not familiar with writing Lua libraries in C yet, although I am interested in learning :)

marcelstoer, 2018-06-23:
I encourage you to give it a try. There's some documentation at https://github.com/nodemcu/nodemcu-firmware/wiki/%5BDRAFT%5D-How-to-write-a-C-module.

TerryE, 2018-06-23:
The slight complication is controlling where the FP routines go.  This is a bit complicated and not for someone developing their first C library.  Let me take a first look at this. :smile:

TerryE, 2018-06-23:
I've just done a quick check: adding the full maths library only hits the flash image size and adds about 18Kb.  There is a small increase in the size of `lmathlib.c`, but the main increase is from the extra `libm.a` routines that get pulled into the firmware.  18Kb is large enough not to do by default.
 
You don't really need to write a new module; just uncomment the functions in `lmathlib.c` that you need and don't forget the corresponding `math_map` entries, then remake your binaries.

I am a bit in overload at the mo, so this is one for you to tackle yourself for now. Sorry.

marcelstoer, 2018-06-24:
> You don't really need to write a new module; just uncomment the functions in `lmathlib.c` that you need and don't forget the corresponding `math_map` entries, then remake your binaries.

Of course, by linking to the draft tutorial I just wanted to encourage @alexandruantochi to add the necessary `#if defined(USE_FULL_MATH_LIB)` so we can have this as a `user-config.h` option. Thanks for pointing that out.

TerryE, 2018-06-25:
I do think that there is an argument for having a subset of the maths functions as standard because we've had previous requests for being able to do simple trig previously.  The minimum practical subset is sin, cos, atan2, exp, log, as these are the one that would be mostly used and rest can be caluated by simple computation.

TheNeurOne, 2018-07-17:
Hi, 
i'm not verry good with such technical english.
I need log function to configure and use a thermal probe with an ESP8266 module (prog in lua of course)

If a understand what you say, i have to call a 'C' lib in my lua code ?
but how to do that ?

TheNeurOne, 2018-07-18:
ok thx, sorry.

ildar, 2018-07-18:
ср, 18 июл. 2018 г. в 4:58, Terry Ellison <notifications@github.com>:

> @TheNeurOne <https://github.com/TheNeurOne> Silvain, I understand your
> question, but this is the wrong place to ask "how do I" questions. Try
> Stackoverflow.com.
>
There are also a few Matrix/IRC/Gitter channels with people in. E.g.
https://riot.im/app/#/room/#esp8266:matrix.org ,
https://gitter.im/nodemcu/nodemcu-firmware and https://webchat.freenode.net/

closed 2019-08-04 by stale[bot]