nodemcu/nodemcu-firmware #2404: Update esptool.py

TerryE, 2018-06-22:
### Missing feature
`esptool.py` is now fully supported by Espressif on is own [git repo](../../../espressif/esptool).  The latest version has some _really_ nice features, including:

-  Sensible automatic defaults such as chip detection and flash size detection.
-  An inflate decompressor allowing esptool.py to zip binaries for downloading.  This makes a _big_ difference to dowload times (I now get over 600 kbit/sec to my Wemos devices).
-  Proper hash-based validation of downloads.

### Justification

Trivial quick win unless anyone can think of gotchas.

### Example image flash

```
esptool --port /dev/ttyUSB0 --baud 460800 write_flash -fm dio  0x00000 $BIN/0x00000.bin 0x10000 $BIN/0x10000.bin;rm screenlog.0; screen -L  /dev/ttyUSB0 115200
esptool.py v2.4.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:d7:1c:89
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0240
Compressed 33456 bytes to 23979...
Wrote 33456 bytes (23979 compressed) at 0x00000000 in 0.5 seconds (effective 495.1 kbit/s)...
Hash of data verified.
Compressed 507904 bytes to 282940...
Wrote 507904 bytes (282940 compressed) at 0x00010000 in 6.4 seconds (effective 631.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
### Workarounds
N/A; current version works

marcelstoer, 2018-06-22:
I would still like to see it removed but _if_ we keep it then should also ensure to keep it up-to-date.

TerryE, 2018-06-22:
We use esptool.py in our Travis builds so even if we get rid of it from `tools`, then we still need to pull or copy it.  A couple of years ago, there was some justification for our keeping our own fork because the Espressif version wasn't being maintained and this gave us the option to fix issues.  However this situation has flipped, and ours isn't maintained and the Espresssif master now has a lot of nice extra features. so my vote is to kill our obsolete copy.

marcelstoer, 2018-06-22:
> my vote is to kill our obsolete copy.

Hmm now you've got me confused because the title says "Update esptool.py" 😲 You mean you want to update it by removing and submoduling it?

Back in the #595 days Johny created a trimmed down version that only contains the image-building features: https://github.com/DiUS/nodemcu-firmware/commit/0ee772312fe5179653f07782fb437c2d467afd07#diff-79af9fd173517961c7b9d67a4600b017 It would reduce complexity if we could just replace our copy with this, adjust the build script and remove the `flash` target from the Makefile.

TerryE, 2018-06-23:
If you look in our top level make we have a rule `$(TOP_DIR)/cache/v$(SDK_FILE_VER).zip:` which simply gets the relevant SDK from the Expressif git repo.  We not just use exactly the process to fetch the equivalent `esptool.py` and use this instead, and delete our `tools` version.

marcelstoer, 2018-06-23:
Sure, it's a little less stable (_remote_ dependency plus "API" that might change someday) but quite ok, I guess.

TerryE, 2018-06-23:
>  it's a little less stable

Not really. Just like the SDK we could fix the pull to a specific git commit for a given master version and its dev successors.  e.g. 
```
ESPTOOL_VER = a7c18f5180ddbc3bf5fc2b7358796159f0f01d18
...
wget --tries=10 --timeout=15 --waitretry=30 --read-timeout=20 --retry-connrefused "https://raw.githubusercontent.com/espressif/esptool/$(ESPTOOL_VER)/esptool.py" -O $@ || { rm -f "$@"; exit 1; }
```

TerryE, 2018-12-31:
Just to follow up my last comment in #2599,  this Espressif version has a lot of nice goodies that are'nt in our obsolete cut-down tools version, including the extra commands:
-  **read_flash_status**. Read SPI flash status register
-  **write_flash_status**. Write SPI flash status register
-  **erase_region**. Erase a region of the flash

and options:
-  **--chip**. Target chip type either esp8266 or esp32 \(defaulting to autodetect)
-  **--before**.  What to do before connecting to the chip \(default_reset,no_reset,no_reset_no_sync)
-  **--after**.  What to do after esptool.py is finished \(hard_reset,soft_reset,no_reset)
-  **--no-stub**. Disable launching the flasher stub, only talk to ROM bootloader. Some features will not be available.
-  **--trace**.  Enable trace-level output of esptool.py interactions.
-  **--override-vddsdio**. Only relevant to ESP32, overriding the VDDSDIO internal voltage regulator.

In the default mode, this loads a 32Kb bootstrap into ICACHE RAM which runs directly from RAM without mapping the flash ROM. This bootstrap code \(see [here](https://github.com/espressif/esptool/tree/master/flasher_stub)) allows the loader to do all sorts of goodies such as:
-  double buffering the downloads so flash write is overlapped with download
-  deflate compression of the downloaded image.  This plus double buffering pretty much halves download times.

The host code can use the stub to do other stuff such as setting the flash size in the `SPIFlashInfo`.

jmattsson, 2018-12-31:
Since the official esptool.py now handles the entry point nicely, it sounds like we should just submodule it. That way it's also version locked, and it brings us closer to getting ESP8266 and ESP32 support into the same tree. I have no particular attachment to our in-tree copy, and I'm not aware of it being more bug-free than the official version. It's served its purpose, but all things must come to an end :)

galjonsfigur, 2018-12-31:
I think @jmattsson idea of adding `esptool.py` as a submodule just like `u8g` and  `u8g2` would be the best option. Also auto flash size detection feature can be used to create `make flash` Makefile target that will work with all ESP8266/8285 modules instead of `flash4m` and `flash512k` targets.

TerryE, 2019-04-05:
Implemented as part of #2692.  The current stable version of the Espressif tool is fetched into `toos/toolchains` as part of the make.

closed 2019-04-05 by TerryE