nodemcu/nodemcu-firmware #2405: nodemcu 里的http有个bug，返回二进制时里面有个\0的字节，它就结束了，其实后面还有的

etimes2008, 2018-06-22:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and delete the rest.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

### Actual behavior

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua
http.request("http://www.dayin.la/attachment/thumb/2018-01/23/user_89580_100x100.jpg", "GET", "Range: bytes=0-100\r\n", "",  function(code, data, headers)
    if (code < 0) then
      print("HTTP request failed")
    else
      print(code, #data, string.byte(data), crypto.toBase64(data), headers["content-length"])
    end
  end)
```
data size must be is 101,but size now is 4
d2ZyAGFmZApzZGY=206	4	255	/9j/4A==	101
### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.
nodemcu-firmware 2.2
### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

anod221, 2018-08-17:
兄弟，我也遇到了这个问题，我修掉了这个bug。你需要的话在群里找我（唯笑不哭）。

translation: 
Hey boy. I met this issue too. Now I fix it, you can ask for the firmware in the qq group if you need it. My nickname is 唯笑不哭

marcelstoer, 2018-08-18:
@anod221 can you please describe the fix or even better create a pull request here?

anod221, 2018-08-19:
@marcelstoer I did these modification.
1 change type http_callback_t in file app/http/httpclient.h.
typedef void (* http_callback_t)(char * response_body, int http_status, char ** full_response_p, int body_size);
2 calculate the body size in function http_disconnect_callback in file app/http/httpclient.c.
3 modify the function http_callback in file app/module/http.c. use lua_pushlstring instead of lua_pushstring.

The original source is hard to modify. So I didn't process some code which may cause the memory leak.
This is why I didn't show the code. You can find the file here.

https://github.com/anod221/nodemcu-explore/commit/1f2665d058fd57723023a48705139fce687f45f7

marcelstoer, 2018-08-19:
@pjsg you were the last to work on the HTTP module, wonder what you make out of this. I'm certainly not in a position to process this.

TerryE, 2018-08-19:
@marcelstoer, I had a quick scan of the code for the module `http.c` and the http library module.  Lots of strlen() calls and `lua_pushstring()` not `lua_pushlstring()` calls so the module assumes Cstring responses.  Will need work to support binary content.  Some of Vowstar's early work.

In the meantime, those that need a workaround can do this in native Lua as per my [HTTP_OTA.lua](https://github.com/TerryE/nodemcu-firmware/blob/dev-LFS-doc-updates/lua_examples/lfs/HTTP_OTA.lua) example.  This also has the advantage that this algo is more stable for non-trivial files since it uses flow control to stop the sending server overrunning the ESP.

HHHartmann, 2019-08-15:
Wer should at least mention this in documentation. I didn't get if the modifications by @anod221 introduced the memory leak but we should still check to move it here.

HHHartmann, 2020-05-29:
Fixed with #2985

PR #2985 (2020-05-10): Fix binary and chunked HTTP downloads. Add diffs from @anod221.
Fixes #2405.

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [ ] I have thoroughly tested my contribution.

Checking the sources that @anod221 gave I created these diffs.
They look ok but I did not have a chance to test them yet.

They also fix an issue with chunked encoding replies which I also have not tested yet.
closed 2020-05-29 by HHHartmann