nodemcu/nodemcu-firmware #2407: The user_start_trampoline() code can overwrite application code

TerryE, 2018-06-22:
### Expected behavior

The `user_start_trampoline()` code shouldn't overwrite applications code during first boot. 

### Actual behavior

The `0x0000.bin` contains a magic header word of the form 0xE9 0x03 0xMM 0xSS where MM = 02 for flash mode `dio` and SS is the Flash size byte.  This flash size is computed during the esptool's `elf2image` generation, but can be updated during the `write_flash` operation.  The default size in the `tools/esptool.py` version defaults the flash size to 512 Kb.

The SDK uses the size byte to load config data from the 4th last flash page based on this size (e.g. at 0x7C000 in a 512 Kbyte flash image) during boot.  If this data is missing then the SDK dies in an infinite loop.  So the trampoline code uses this size byte to validate that the config data is correctly initialised and if not then rewrites this.  However if the actual application code is large enough to include this incorrect flash page then this function will stomp over the actual firmware code, corrupting the image.  

So this failure only manifests if:

- The flash image size byte denotes a 512Kb flash, and 
- The 0x10000.bin image is larger than 0x6C000 (442,368) bytes

### Test code
N/A

### NodeMCU version
Latest dev

### Hardware
N/A

### Workarounds

This bug only occurs if the flash size byte is at the default 512Kbyte and large images are loaded.  This can be avoided either by:

-  Using the current Espressif version of the tool as this probes the flash to size the flash correctly, or
-  Use the `-fs` option to explicitly set the flash size.

devsaurus, 2018-06-28:
As Johny mentioned in the LFS patch discussion, the FW could handle this corner case if the size byte fix-up is moved to the trampoline code.
This improves overall fail-safe operating at the expense of allocating valuable IRAM.
Should we trade in IRAM for covering a corner case that can occur just once during the life cycle and is caused by an inconsistent flashing configuration?

TerryE, 2018-06-29:
When you look at the generated code, we could handle the corner case in roughly the same code footprint of we recoded it, e.g. replaced to case cascade by a byte map lookup, and did a single while loop. 

Once we done the current LFS PR, I'll take a quick look and post the code fragment here.  If we did this, then we could also drop the later resize.  If you are happy then I can raise the PR.

TerryE, 2019-08-10:
This should have been closed anyway, since this is no longer an issue with the new SDK 3.0 Partition Table.

closed 2019-08-10 by TerryE