nodemcu/nodemcu-firmware #2408: RFC:  Views on how to implement host-side provisioning

TerryE, 2018-06-23:
I use my luaOTA provisioning system, but few others do.  One of the main PITAs, IMO is that it requires a host-side Lua installation and for whatever reason most NodeMCU developers seem reluctant to do this.  So I want to remove this requirement.

I am interested in views on possible approaches for doing this.  I see there are two primary alternatives to my doing a minimal change and leaving this largely as a personal tool:

-  Add the Lua File System and MD5 libraries to the `app/lua/luac_cross` build so that you can use the `luac.cross -e` option to execute the provisioning system.  Having these has other uses: for example if you want to move small text files such as `js`, `html`, etc. into a resources module were these are string constants which can then be loaded into LFS, then having the LuaFS makes doing this as a `-e` script fairly straight forward as well.

-  Adopt the `esptool.py` approach and reimplement the server in Python.  It's a decade since I did any Python coding, but I am sure that I can rapidly dust this off.  But I do this that this is a sort of "give up" statement if we end up writing server-side code in Python when Lua is equally as good.

I will listen to any views and close this issue in a few weeks.

Marcel, I realise that this doesn't fit the template but I think that the committers who end up closing most issues deserve a few liberties. :smile:

TerryE, 2018-06-23:
@jmd13391 Joe, feedback from active users like yourself is very useful in helping the developer group focus their priorities.

IMO, a real concern with IoT devices is security -- especially if they are provisioning over the public internet. My personal approach to this is to isolate my own IoT devices on a control WiFi service and to use a locked-down RPi3 (clone) acting as a local host and bridge to any public services.  My IoT devices cannot directly access the Internet and conversely any Internet access to them or their data is through properly authenticated and trusted connections.

I realise that this might not be a viable approach for specialist IoT suppliers, but policies such as the Internet Society's [IoT Trust Framework](https://www.internetsociety.org/iot/trust-framework/) are essential, IMO but are still ignored by many mainstream IoT developers.

TerryE, 2018-06-23:
Given that that the compile and rebuild cycle for LFS is **_so_** short (a couple of seconds to compile and reload and reload an absolute LFS region + the new `LUA_INIT_STRING` config directive:
```Lua
#define LUA_INIT_STRING "pcall(function() node.flashindex'_init'() end)"
```
or a function variant with a SPIFFS fallback if the LFS isn't loaded such as
```Lua
function() local f,a=node.flashindex'_init'; return a and dofile'reinit.lua' or f() end
```
so that if the LFS isn't loaded, then the startup falls back to executing the SPIFFS file `reinit.lua`.

I am now planning replacing my prod apps by a fully LFS packaged one, where SPIFFS is only used for volatile configuration and other application data and a recovery lua fallback in case there is a powerfail during LFS update. 
 
In this case the application update now reduces to downloading a new LFS image and executing a flashreload.   The one wrinkle that I was considering adding to the reload was to make the image file compressed and to uncompress this during reload.  This would allow us to do an integrity / hash check so that only valid LFS would be flashed.   You could pull the image from any suitable HTTPS cloud service if you want.

joysfera, 2018-06-26:
Not sure if my amateur OTA implementation can add anything useful here but I'll explain it anyway. I prepare the *.lc files on one of my development ESP8266 (using esptool that uploads *.lua and compile on the ESP), then download them from the dev ESP and upload them to the provisioning server with apache web server. Later I'll let the ESP8266s in the field download the *.lc files using plain http get. This might sound cumbersome at first but I basically just mirror a working setup from one device to all others. Thus I don't need any Lua on the server, and I can use any public web hosting that offers downloading plain files (all do).

When I switch over to LFS I'll simply replace the bunch of *.lc files (that often fail to download completely) to a single LFS image that I am again prepared to prepare separately on my development ESP8266 first and then move on the public webserver.

So maybe this can inspire @TerryE a bit. Lua on the server might not be such issue...

TerryE, 2018-06-27:
@joysfera, I assume that implement your own http stack in Lua rather than use the `http` library as the latter can't cope with large content gets.  Have a look at my ftp example where I do flow control to prevent the server overrunning the ESP8266.  

As I said, I am coming around to a keep-it-simple-stupid approach to provisioning lfs.img files.

nwf, 2018-06-29:
I also have a KISS "provisioning system" (if one will be generous with their vocabulary) that uses the little mockup of a telnet server I wrote to write files (often, lua source rather than cross-compiled byte-code) to the spiffs.  It's all driven with expect around netcat host-side: https://github.com/AcmeTensorToys/esp-lua_core/blob/master/host/pushvia.expect, which, because it's waiting for prompts from the nodemcu device, is sort of naturally rate-limited.  The server itself is https://github.com/AcmeTensorToys/esp-lua_core/blob/master/telnetd/telnetd.lua but loads https://github.com/AcmeTensorToys/esp-lua_core/blob/master/telnetd/telnetd-file.lua to do the actual fs work.

I, too, have been meaning to replace this whole thing with @TerryE 's OTA world, but if another example of "what other people do" is useful, there 'tis.  If not, please pardon the noise.  FWIW, I have no objection to having a host-side lua environment (and, in fact, often prototype my nodemcu applications' logic under test harnesses host-side).

TerryE, 2018-06-29:
See #2315 for discussions on luac.cross.  This is all pretty easy for Linux and Mac users; a minor hoop for Win10 Pro users and a PITA for all other Windows users that don't want to run a Linux VM as we've got no WinX developer willing or with the skills to do the make on something like MinGW to build a WinX executable for those developers that want to use this.

TerryE, 2018-07-07:
I have decided to go "thin server"; that is to use a standard webserver as the host-side delivery vehicle and drop the host-side Lua server entirely.  Thanks for the discussion and feedback.

closed 2018-07-07 by TerryE

TerryE, 2018-07-07:
@develperbra, I'll be pushing an LFS based OTA solution within the next week.  :)