nodemcu/nodemcu-firmware #2409: Addition of compression / decompression libraries

TerryE, 2018-06-25:
### Missing feature

We should add some form of lossless compression / decompression library  and module to ESPs. 

### Justification

Most compression strategies can reduce data sizes by typically 40% (+/-).  This can be a material saving for transferring data over networks in terms of traffic, latency, etc. 

### Workarounds

On most IoT applications, data can be compressed on-host and stored in compressed format on SPIFFS, etc., so that HTTP delivered static components can be delivered compressed without the need to have on-ESP compression libraries,

However this doesn't work for sending compressed data to the ESP as this will normally require on-device decompression.  It is also a problem if the application wishes to compress realtime data.

### Issues and constraints.

It isn't sensible to try and develop our own algos, but should base this on an already proven implementation.  However, we also have to avoid encumbering the firmware with restrictively licensed code, so any libraries should be patent-free and MIT licensed (or more permissive).  A possible alternative might be a BSD library with any  wrapper code original and NodeMCU licensed.

There is a general trade-off between the time+complexity of the compression algorithm and performance, and this is a major issue on memory limited IoT devices.  I split such algos into three broad categories: **Lightweight**, **Standard** and **Advanced**.

-  **Standard** algos are the _deflate_ and _gzip_ algos used for HTTP content compression.  Both are based on the LZ77 algorithm + Huffman coding.
-  **Lightweight** algos typically target embedded and equivalent use where decode performance and memory use are important constraints. Hence the typically drop Huffman coding and use restricted window  LZ77 algorithms.
 -  **Advanced** such as `bzip2` and `xz` allow the luxury of higher compression time and memory resources to achieve higher compression performance; however their implementations are outside the resources available on an ESP8266 processor and so I discount these from further discussion.

The `esptool.py`  uses a deflate-based implementation for on-the-fly compression of image downloads to the ESP.  This typically achieves ~40% compression in the case of firmware images and this can be far higher in the case of SPIFFS file system images.  Also there are numerous free downloadable host utilities (WinX, Linux, etc.) that can encode or decode such streams.  However this runs as a bootstrap on a bare CPU, rather than co-existing alongside a NodeMCU + SDK build.  So resource usage must remain a risk here.

Most lightweight implementations drop the Huffman encoding (or use a very restricted static coding) and use some variant of LZ77.  Their decoders will typically run fast  and with minimal code and RAM requirements, and so are well suited to IoT applications.  However PC-side conversion is an issue so if we go this route then the corresponding command line utilities should be downloadable, either from the original project or as part of our tools make.

My instinct is to stick with a lightweight implementation, but I would be interested in the view of other contributors, e.g. @jmattsson @pjsg @devsaurus @marcelstoer @djphoenix,

TerryE, 2018-06-25:
Just as a codicil, my intent would be to provide a wrapper encode / decode library, plus `luac.cross` support for both and that file image forms of LFS compiles would encoded using this algo.  Compressed LFS images would also be MD5 signed, so that the image signatures could be validated before any reflash was executed.

jmattsson, 2018-06-25:
Have you looked at https://github.com/pfalcon/uzlib ?

TerryE, 2018-06-25:
Yup it was on my list, but @jmattsson what do you think of the basic approach?  Now that the latest LFS patch supports `LUA_INIT_STRING` properly, I can strip the reprovisioning model back to downloading a new copy of the LFS image and reflashing the LFS.  I suggest that we have three levels of compression support selectable: none, inflate only, both (though the `luac,cross` image would always include both.)

TerryE, 2018-07-02:
> Have you looked at https://github.com/pfalcon/uzlib ?

I have these working fine with some rework - the compression code is table driven with a lot of static initialised tables, which is bad news, so I added a simple fast generator init routine to build these in malloced memory and this works fine.  

They are fast and small, and can decode any `gzip` compressed stream and the output compressed stream is likewise readable by `zcat`.  To keep things simple, the compression utility uses static Huffman trees, which give good place for small records, but in the case of larger data streams such as a `luac.cross` output this will typically give 30% side reduction as opposed to 40% with standard compression which builds an optimised tree.  

 I want to wrap these in a module to give Lua API support, as well as integrating them into LFS compile and reflash.  This latter requires adding stream support for input and output.  So whilst straightforward, it's also a bit of work, so this is lower down my TODO list.

TerryE, 2018-07-06:
Just another progress update.  As Johny suggested, I am basing this work on @pfalcon's  [uzlib](https://github.com/pfalcon/uzlib). I am trying to keep the code compact and fast, so I am using a subset of this code:
-  The code makes a lot of use of constant tables to drive the compress / expand.  Our make puts these into flash, and we use a S/W exception handler to catch and implement 8 and 16-bit accesses, which slows down execution.  I've replaced these by mallocing RAM for the tables for the scope of the operation and using some init routines to create these dynamically.  This works out a lot better in terms of runtime and overall RAM usage. 
-  In particular the decompress has to be able to expand a SPIFFS file of say 160Kb into 256Kb flash, but doing so within the RAM constraints of the ESP, so we need a streaming expander which can run with a buffer space of ~16-32Kb.  The default compress uses a 32Kb sliding window as a dictionary.  Dropping this to a 16Kb window only has a small impact on compressed sizes, but almost halves the RAM needed.
-  The code is variants of the permissive ZLIB license and so it compatible with our MIT one.
- I will add a host-based compressor, as most host tools don't allow you to select a 16kB dictionary.  (No host base expander is needed as the data stream is zlib compliant.)  This will also allow me to evaluate trad-offs for window size on typical LFS images.

TerryE, 2018-07-24:
I am in the process of stress testing the addition of my rework of Pual's uzlib.  In the end I had to rewrite his compression and decompression cores because the original versions took up far too much statically allocated RAM.  I get roughly a 3× compression ratio, so the `luac.cross` LFS image for a maxed out 64 Kb region is typically just over a 20Kb download.  Incidentally dropping from a 32Kb window to a 16Kb one makes less than 1% difference in compressed image sizes.

I will also add a **Lua** example of a re-provisioning bootstrap which can download and reload an LFS image from a standard web-service.

jmd13391, 2018-07-25:
"_I get roughly a 3× compression ratio, so the luac.cross LFS image for a maxed out 64 Kb region is typically just over a 20Kb download_" -- 🥇 

"_I will also add a Lua example of a re-provisioning bootstrap which can download and reload an LFS image from a standard web-service._" -- 🥇 🥇 

@TerryE - Simply outstanding... We are not worthy.  :-)

-- Joe

TerryE, 2018-08-02:
OK, my latest PR #2448 uses this library: `luac.cross:lflashimg.c` calls the deflate and the LuaVM `lflash.c` calls the inflate to write to flash.  Paul Sokolovsky's, (@pfalcon) deflater version required the whole input to be in memory, and it allocated a lot of the reference tables statically and hence hogged valuable RAM, so I in the end it was just easier rewriting it from scratch and the RFC.  My deflater is a bit faster and typically achieve 3× rather than 2× compression.  This being said, I must give Paul Sokolovsky my thanks because his minimal library was an excellent starting point.

The inflater is designed to explode files far larger than RAM to flash, and because it can't guarantee that it is run immediately after boot, the heap might already be defragmented, so it allocates its working memory in 2K pages.  The RFC stipulates a dictionary window of 32K but this would require ~38Kb RAM free in 2K pages which would kill us on an ESP8266, so I use a 16K dictionary (which drops the compression less than ½%, so big deal).  When coupled with my new `HTTP_OTA.lua` script this happily reloads the LFS from a webserver in about a second.  The slowest part is waiting for the ESP to reboot.

So all looks good.

At some stage I will add a library wrapper so these inflate and deflate routines are directly callable from Lua.

AllXXX, 2019-01-08:
https://archive.codeplex.com/?p=zipstorer

TerryE, 2019-01-08:
or even or own `app/uzlib` :rofl: Why add another library especially one that we'd have to migrate from c# ?

closed 2021-06-20 by stale[bot]