nodemcu/nodemcu-firmware #2412: [ESP32] Handle json encode/decode

Lir10, 2018-06-28:
Looks like sjson is not supported in esp32 nodemcu firmware

I'm trying to migrate an app from esp8266 to esp32 but without sjson module it's pretty much useless

Any ideas about this?

x-logLT, 2018-07-29:
You can pretty much take the sjson module/lib from esp8266 branch and add it to ESP32 one without much trouble, have been using it as such close to a year now: https://github.com/x-logLT/nodemcu-firmware/commit/9b003b55b1355d12a9042227ddc8bea4d367575a.
Other way would be using included cjson lib and adapting old cjson module to it, don't know how straightforward that would, if memory serves right, its not even the same lib as was used in old esp8266 branch.

marcelstoer, 2018-07-29:
>  have been using it as such close to a year now: x-logLT@9b003b5

Great, then why not put it up as a PR to help the community?

x-logLT, 2018-07-30:
it's on my to-do list, at the moment I don't have time to do it the proper way.

galenzhao, 2018-09-27:
by using this 
https://gist.github.com/tylerneylon/59f4bcf316be525b30ab

x-logLT, 2018-11-17:
So which route are we taking;
1. Use masters SJSON lib/module
2. Use esp-idf's cjson lib and write module for it
3. Use LUA module based on lib pointed out by galenzhao
?

devsaurus, 2018-11-17:
I'm unbiased here. Are there any pros or cons with either of the first 2 alternatives?

I understand that support for streaming JSON is essential, though.

jmattsson, 2018-11-17:
I'd suggest going with the sjson option. Not only is it built for streaming support, it also means we're not diverging further from esp8266 nodemcu.

devsaurus, 2018-12-23:
Related PR was merged.

PR #2562 (2018-11-22): ESP32: Add SJSON module
Fixes #2412 

- [x] This PR is for the `dev` branch rather than for `master`.
- [x] This PR is compliant with the [other contributing guidelines](https://github.com/nodemcu/nodemcu-firmware/blob/dev/CONTRIBUTING.md) as well (if not, please describe why).
- [x] I have thoroughly tested my contribution.
- [x] The code changes are reflected in the documentation at `docs/en/*`.

This adds SJSON module taken directly from master.
closed 2018-12-23 by devsaurus