nodemcu/nodemcu-firmware #2414: RTD not building

marcelstoer, 2018-06-29:
RTD silently updated their MkDocs version to 0.17.3. This breaks the docs build

> WARNING -  Config value: 'theme_dir'. Warning: The configuration option {0} has been deprecated and will be removed in a future release of MkDocs. 
> 
> Aborted with 1 Configuration Warnings in 'strict' mode!

Currently our `rtd-requirements.txt` in the root is inactive. Analyze if we still need it.

marcelstoer, 2018-06-30:
All good now

closed 2018-06-30 by marcelstoer

marcelstoer, 2018-07-01:
I noted the new section, thank you for that, makes sense to me. You had the LCD/LFS labels pointing to the respective other. I fixed that in 3f396cba22ebeb0a202a93ffc614cdbdacf54f68.

For the record, the RTD issue: rtfd/readthedocs.org#4314

TerryE, 2018-07-01:
> You had the LCD/LFS labels pointing to the respective other. 

Durrrhh. Typical Terry mistake. Thanks for this.