nodemcu/nodemcu-firmware #2415: Issue with bme280.qfe2qnh wrong arg range

alexellis, 2018-07-01:
Make sure you read and understand http://nodemcu.readthedocs.io/en/dev/en/support/.
Use one of the two templates below and **DELETE THE REST**.

Hi, I'm not sure if this is the right place to raise the issue, but it's where I can see the source-code and given that this is flashing the firmware built from this repo I was expecting it and the examples to work with the device. 

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

Expected this to work (mentioned in docs page for bme280)

```
bme280.qfe2qnh(Q, alt)
```

### Actual behavior

```
stdin:1: wrong arg range
stack traceback:
        [C]: in function 'qfe2qnh'
        stdin:1: in main chunk
> 
```

### Test code
Provide a [Minimal, Complete, and Verifiable example](http://stackoverflow.com/help/mcve) which will reproduce the problem.
```Lua

alt=217 -- altitude of the measurement place
mode = nil
mode = bme280.init(4, 5)

local alt=39 -- altitude of the measurement place

mode = bme280.init(3,4)
print(mode)

P, T = bme280.baro(alt)
print(string.format("QFE=%d.%03d", P/1000, P%1000))

> QFE=1014.599

QNH = bme280.qfe2qnh(P, alt)
print(string.format("QNH=%d.%03d", QNH/1000, QNH%1000))
```

### NodeMCU version
Which branch are you on? If you know the Git revision then add it here as well.

Via cloud builder:

> 1.5.4.1-final (frozen, for 512KB flash)  

[image]

### Hardware
Describe which ESP8266 device you use and document any special hardware setup 
required to reproduce the problem.

NodeMCU dev board

Detecting chip type... ESP8266
Chip is ESP8266EX

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

vsky279, 2018-07-20:
First it's good to read the instructions when submitting and issue. The parts in __bold__ are especially usefull like e.g. the note "__DELETE THE REST___". :-)

Jokes apart. Does the example really work for you? You should call first `bme280.setup()` function in order to initialize the sensor.

Why are you passing a parameter to `bme280.baro()` function? This function has no parameters.

Try to call `bme280.qfe2qnh()` directly with numerical parameters like `bme280.qfe2qnh(1014599, 39)`. Do you get the same error?

closed 2018-07-20 by marcelstoer

alexellis, 2018-07-21:
> First it's good to read the instructions when submitting and issue.

Being a maintainer is hard and it's frustrating when you think someone is deliberately not following your process - communication is also hard over the internet. I'm not perfect, but this came across less than friendly. I read the instructions and wrote out all the required details. As a first time contributor I didn't want to delete the template in case you asked for some edits.

> Jokes apart. Does the example really work for you? 

No it doesn't work, it produces the error above which is why I raised the issue. This example is taken from the NodeMCU docs so I found it very confusing.

> Also please note that the function bme280.init() has be deprecated in the development branch already. For future compatibility you should you i2c.setup().

Deprecated in which version specifically? It's unclear which version to use from the cloud-builder - I have shared above the version used (`1.5.4.1-final (frozen, for 512KB flash)`

This is the resulting blog post and it contains the code shared in a Gist. (I found a way to make this work and that meant using `init`)

https://blog.alexellis.io/iot-nodemcu-sensor-bme280/

Thanks for the additional information.

marcelstoer, 2018-07-21:
@alexellis congrats on a great blog post! I like such clean sites.

You chose to use an old stale branch. My cloud builder hints as much ("final", "frozen") and the branch's README warns about this prominently
> ! This branch is frozen at the last commit before the Espressif SDK was upgraded to 2.0 !

The most appropriate answer probably would have been to state again that we won't change frozen branches. Not even bug fixes are back-ported.

Your statement

> This example is taken from the NodeMCU docs so I found it very confusing.

refers to https://nodemcu.readthedocs.io/en/1.5.4.1-final/en/modules/bme280/ and I guess you're right about this. If you look at the documentation of the current master & dev branches e.g. http://nodemcu.readthedocs.io/en/latest/en/modules/bme280/ you'll notice that `init` is gone.

As you seem to be using a [1st gen NodeMCU module](https://frightanic.com/iot/comparison-of-esp8266-nodemcu-development-boards/#v1) (4MB flash) there's actually no reason IMHO to use the 1.5.4.1 branch.