nodemcu/nodemcu-firmware #2418: Wording for README.md

TerryE, 2018-07-02:
Another issue that doesn't fit the templates.  This is a trivial one but worth discussing and reaching consensus before raising the PR.  There are a could of issues with the project README that deserve fixing.  \(Many thanks to @nailujx86 for raising this\).

-  The lede is just misleading and needs replacing.  For example: it majors eLua whilst in reality perhaps 85% of the Lua hierarchy is standard Lua. Of the remaining mods and new code, my and zeroday's mods / enhancements are the eLua components are a small percentage (mainly ROTable handling) and

marcelstoer, 2018-07-02:
Fine for me, I like. Two issues:

- Don't mention the exact NON-OS SDK _version_. Would be one less place to edit with every SDK upgrade.
- "can now be run on any ESP module." - I don't think the term "ESP module" has any official meaning.

How about 

> ...nity-supported. The firmware can now be run on any ESP8266, ESP8285 and ESP32.

-> maybe mention something about the respective NodeMCU branches?

cwrseck, 2018-07-03:
It would be helpful to add the release version (eg. 2,2,0-master_20180608) to the title of the README title.  I have several versions of NodeMCU, and it's tedious to have to check the source directory to find out which version's documentation I've found.

Will

TerryE, 2018-07-03:
@cwrseck.  I can't follow your logic here.  The README is in the source directory.  How do you do your build? Via cloud builder or what?  From our PoV, having the version in the README is a PITA as you now have to update this file every release.

cwrseck, 2018-07-04:
I have a docs link on the desktop, effectively (I'm building NodeMCU from a local directory).  With various versions in play, it's handy to know which version of the docs goes with which without scuffling through the file system.   In general I'd expect the version number to be updated automatically when each new release is built,  but there may well be difficulties I'm not aware of.

Will

marcelstoer, 2018-07-20:
> worth discussing and reaching consensus before raising the PR

Ready for PR IMHO.

marcelstoer, 2018-07-24:
Fixed in 8505362

commit 8505362f (2018-07-24): Rewrite intro for README

Fixes #2418
closed 2018-07-24 by marcelstoer